benaclejames/VRCFaceTracking #3: Parameters no longer update.

Azuuth, 2020-11-07:
First of all thanks a ton for making this! It's the one good use I've gotten out of my Pro Eye's tracking hardware. 

Running Melonloader v0.2.7.2 with VRChat version 2020.4.1p5, build 1010. 

Despite this being the update that was supposed to fix parameter sync, the following parameters no longer update based on my eye's tracking positions, either locally or to other players. All stay at their default values of zero: 
-LeftEyeX
-LeftEyeY
-RightEyeX
-RightEyeY
-RightEyeLid

I'm not currently using any other tracking parameters on my avatar so haven't been able to test them in game, but I suspect they won't update either. I've attached my most recent MelonLoader log to this report. If you need any more info or 
help with testing, I'm Azuth #1016 on discord or just Azuth in VRC. 

[MelonLoader_20-11-06_18-47-26.730.log](https://github.com/benaclejames/VRCEyeTracking/files/5504224/MelonLoader_20-11-06_18-47-26.730.log)

benaclejames, 2020-11-07:
Hey!
Thanks for the report. Currently, I'm working on a small patch to make the mod compatible with the more current versions of VRC due to the way avatars are spawned changing slightly. I've attached a "bodge" version that will work for the time being. It's not perfect but it works... I'd still recommend waiting for the patch though 😆

[VRCEyeTracking.zip](https://github.com/benaclejames/VRCEyeTracking/files/5504233/VRCEyeTracking.zip)

Azuuth, 2020-11-07:
Talk about a quick response, thanks for this!

benaclejames, 2020-11-08:
@Azuuth The issue should be fixed with the [latest release](https://github.com/benaclejames/VRCEyeTracking/releases/tag/v1.1.1). Please let me know if things start working again on your end 😄

closed 2020-11-11 by benaclejames