benaclejames/VRCFaceTracking #4: Performance Issue

ShawnMayer, 2020-11-26:
First of all I really appreciate the great mods you made.
Using this mode, I was able to apply eye tracking.

However, the only inconvenient thing is that if you apply this mode, CPU Frametime is very high.

My CPU is 9900k (with overclock to 5.1Ghz) and GPU is RTX 3090.
Nevertheless, the frame is hard to exceed 45 when applying this mode.

I don't think this mod itself has any performance-degrading issues.
I am wondering if it is possible to improve performance by manually disabling the unused one of the various parameters.
If that's difficult or doesn't work, I hope at least you add the ability to easily turn this mod on and off in-game. (Using UIExpansionKit)

Once again, I sincerely thank you for making this mod, and we look forward to improving it further.

ShawnMayer, 2020-11-27:
I found a way to improve the frame while using eye tracking today.
I experienced a temporary error while playing VRCHAT.
Since then, we have noticed that the eye tracking frequency has dropped significantly.
I felt that the eye tracking frequency fell from 120 to about 30, and VRCHAT's frame was improved instead.

If we find a way to adjust eye tracking frequency, I expect it to help improve frame.

benaclejames, 2020-11-27:
SRanipal is quite performance intensive. If you need the extra frames badly, it might be a good idea to turn off the eye tracking. There would be little to no difference in ignoring the unused avatar parameters as the SRanipal SDK returns all data at once regardless.

eslafur, 2020-12-25:
I had a huge performance impact dropping from ~50 frames to 15 and less.
looking into it a friend found out that `MelonLoader.MelonCoroutines` actually runs these threads in the main thread, blocking the game from running.
we are currently working on refining the seperation into a proper thread and I already have no performance impact at all.
running 45 Frames (as without mod) with working eye tracking.

when we are done with the optimized threaded implementation I'll make a pull request to push the fixed version.

benaclejames, 2020-12-25:
Many thanks for the reminder, I completely forgot the difference between Monobehavoir coroutines and system coroutines. I'll be pushing a fix to implement real threading tomorrow along with support for "Squinting" readouts

ShawnMayer, 2020-12-27:
> I had a huge performance impact dropping from ~50 frames to 15 and less.
> looking into it a friend found out that `MelonLoader.MelonCoroutines` actually runs these threads in the main thread, [...]

It sounds great!!!
I'm very looking forward to the release of the fixed version!

benaclejames, 2021-01-19:
Fixed in the latest update

closed 2021-01-19 by benaclejames