benaclejames/VRCFaceTracking #6: Eyes do not track

AmaliaKalio, 2021-01-19:
Release: 1.2.0
VRC Version: Build 1033
DLLs loaded in MelonLoader?: Yes

I've got an Avatar 3.0 package with existing Parameter fields. To test, I've added LeftEyeX, LeftEyeY, RightEyeX, RightEyeY, and LeftEyeLid as Float. EyeLook has been disabled in the descriptor, and Additive shows "None (Runtime Animator Controller)".

Based on the provided documentation, it's unclear if anything else needs to be added to the avatar package. There's mention of simple blendstates, but it's unclear if the parameter items are all that needs to be added or not based on the docs.

benaclejames, 2021-01-19:
Have you added animations for the parameters inside your fx controller? The mod only automates setting the parameters with information from the eye tracking sensors but you still need to define the animation layers and blend trees to move the eyes from the parameters. For example, my eye look blend tree looks like this: 
[image]

AmaliaKalio, 2021-01-19:
This explains much. Thanks!

ehefk, 2021-01-20:
Was this able to get it working?  VRCMods has the plugin listed as "Not working"

benaclejames, 2021-01-20:
Can confirm it works, released a build a couple hours ago that gets it working with the new update but it's not fully complete yet. Still need to figure out network priotization but other than that, should work fine!     https://github.com/benaclejames/VRCEyeTracking/releases/tag/v1.2.1

bukneq, 2021-01-20:
Awesome. I'm also stuck on the blend trees, mostly because I'm not sure what to do with what. Would it be possible to add your .anim(ations) for the blinking and moving, as well as the FX controller? This would help a lot, even if I'd have to adjust them.

AmaliaKalio, 2021-01-20:
> Awesome. I'm also stuck on the blend trees, mostly because I'm not sure what to do with what. Would it be possible to add your .anim(ations) for the blinking and moving, as well as the FX controller? This would help a lot, even if I'd have to adjust them.

Aye, same here. Examples would be great for reference - Unity keeps throwing an error saying that rotational keyframes aren't supported on humanoid rigs. Even if there's extensive edits involved, it would be good to see how it works in at least one case.

benaclejames, 2021-01-21:
Been thinking about making an example avatar for a while now but I'm still struggling to find a good model to use. Once I do find that, though, I'll include all the anim files and layers required. @AmaliaKalio To fix the issue you're having, remove the eyes from the avatar definition on the model

InconsolableCellist, 2021-01-21:
Here's my avatar setup

* Add the parameters you want to use to your VRCExpressionParameters Object.
* In your FX Object, add the parameters and then make layers for the animation. I independently control the eyes and have left eye blinking control both eyes. So I created the layers "Left Eye", "Right Eye", "Left Eye Blink". I've also added "Eyes Dilation" and "Eyes Widen". Set the weights to 1

**For blinking:**

1. Create a blend tree as the default state in the layer. Inside the blend tree set the parameter to LeftEyeLid and add tow Motions, "anim_blink_both" and "anim_unblink_both". (Or similar.) The animations should blink and un-blink the eye(s).

2. Have anim_blink be set closer to 0 and anim_unblink be set closer to 1. I used threshold values of 0.4 and 0.9 respectively, which helped fully control the eyelid.

**For eye movement**

1. Create a blend tree in the left/right eye layer and set it to be 2D Freeform Directional. Set the two parameters to LeftEyeX and LeftEyeY (or RightEyeX/Y). 
2. Add five animations with the following PosX and PosY values:

anim_left_eye_left  -0.7, 0, 1 
anim_left_eye_right 0.7, 0, 1
anim_left_eye_up 0, 0.7, 1
anim_left_eye_down 0, -0.7, 1
anim_left_eye_default 0, 0, 1

Those animations should control the eye blendshape(s) to move the eye(s) in those directions. See the attached screenshots

[image]

[image]

AmaliaKalio, 2021-01-21:
@benaclejames I typically have been keeping "Eye Look" disabled, but went ahead and briefly re-enabled it to remove the eye references from the bone slots before disabling again. No dice, unfortunately.
[image]

Curiously enough, I can edit the .anim file via notepad and force the values in that way, but if I ever scrub through the time in the animation file within Unity, it reverts the rotation.x value back to 0. ¯\_( ͡° ͜ʖ ͡°)_/¯ I guess I'll try doing this for the other directionals on one eye and see what happens.

bukneq, 2021-01-21:
Thanks for the better explaination. Would it still be possible to get those .anim files uploaded? That would help a lot.

InconsolableCellist, 2021-01-21:
The anim files have to be customized for each avatar, and you might have to add blendshapes to your avatar for the eye deflection. Here's an example video on how to create animation files: https://www.youtube.com/watch?v=iYDDVbx5Wpk

You'll have to either create them yourself for your avatar (and assign them in Blend Trees) or commission someone to do it for you. I wrote a more complete guide on setting it up here which may help: https://github.com/InconsolableCellist/VRCEyeTracking/blob/docs_improvement/EXAMPLE.md

If you're looking for someone to set it up for you, you could view my profile and contact me. But without seeing your avatar setup it's not possible to tell you exactly what settings to change, if the above info and guides don't help. I'd like to make a video at some point on how to do it

ehefk, 2021-01-22:
I've had no problem getting blinking to work, but when I go to set up eye movement I'm trying to use a rotation transform to move the eyes and it doesn't seem to be working...debugger shows that they're tracking.

Is the best way to create blendshapes then animate like I did the blinking?

Thanks!

AmaliaKalio, 2021-01-23:
I can confirm that even with blendshapes, it doesn't seem to trigger anything. This also includes blinking, in my case. I have tried where the animation contains two keyframes of different values (IE: Center to position in the case of look animations, or Open to closed/visa-versa for blinking), or where there are two keyframes of the target value, as well as just using a single keyframe. I can confirm in the Animation editor that these _should_ be setting the proper items and values.

The closest I've seen was when the avatar didn't load in right and I couldn't even use standard toggles. At this point, the left eye was stuck looking up. Reloading the avatar resolved both of these, but there's still no tracking. At the very least, it tells me that the parameters are plugged in for the animations.

benaclejames, 2021-01-23:
@AmaliaKalio Do you see the param values changing in the debug menu?
@ehefk You may need to create an avatar mask and remove the eye bones from your avatar definition for the bones to rotate

AmaliaKalio, 2021-01-23:
@benaclejames I do, yes.

ehefk, 2021-01-23:
Thanks Ben!  

I watched a quick video on how to make an avatar mask, selected just the eyes and removed the bones from the avatar's config screen and now have tracking eyes!

bukneq, 2021-01-25:
After setting up everything, the blinking animations as well as the eyes movement work. Unfortunately the eyes barely move, even with adjusted thresholds... Am I missing anything? They do move, but just about nothing, even though the maximum of the animation has been placed further.

AmaliaKalio, 2021-01-25:
@benaclejames Found the issue (ish). The short of it is that it was on my end. Animations that I had created in Unity checked out, but there was something built into the avatar preventing them from triggering properly (likely some sort of animation conflict, though I still haven't been able to track down where/what). Got that sorted by going about it a different way, and can confirm it works as-intended!

Kieaer, 2021-02-02:
Doesn't work parameter change in VRChat build 1040

benaclejames, 2021-02-02:
@Kieaer Are you sure everything is setup correctly on your end? Nobody else has reported any issues and I can't see any issues locally. Are you on the newest release and not in any VRC betas?

benaclejames, 2021-02-06:
Closing due to inactivity

closed 2021-02-06 by benaclejames

Kieaer, 2021-02-07:
Solved in last release. thanks!