benaclejames/VRCFaceTracking #7: Unable to load EyeTrack.dll

Adjerry91, 2021-01-20:
I had the eye tracking working last week. Now it is failing to launch in Melonloader. The TobiiEye tracking is working in the corner and the calibration.

[image]

InconsolableCellist, 2021-01-20:
Same here:

[code]
[22:08:29.960] [MelonLoader] ------------------------------
[22:08:30.188] [MelonLoader] [Error] Unable to load C:\games\steam\steamapps\common\VRChat\Mods\EyeTrack.dll:
System.Reflection.TargetInvocationException: Exception has been thrown by the target of an invocation. ---> System.TypeInitializationException: The type initializer for 'EyeTrack.MainMod' threw an exception. ---> System.TypeLoadException: Could not load type of field 'EyeTrack.MainMod+AV3Parameter:ParamEnum' (1) due to: Could not resolve type with token 01000027 (from typeref, class/assembly EnumNPublicSealedvaStNoSt18StStStStStUnique, ) assembly: type:EnumNPublicSealedvaStNoSt18StStStStStUnique member:(null) signature:<none>
  at EyeTrack.MainMod..cctor () [0x00000] in <e964e39cc49442129fcb72f8cc967183>:0
   --- End of inner exception stack trace ---
  at (wrapper managed-to-native) System.Reflection.MonoCMethod.InternalInvoke(System.Reflection.MonoCMethod,object,object[],System.Exception&)
  at System.Reflection.MonoCMethod.InternalInvoke (System.Object obj, System.Object[] parameters) [0x00002] in <e1319b7195c343e79b385cd3aa43f5dc>:0
   --- End of inner exception stack trace ---
  at System.Reflection.MonoCMethod.InternalInvoke (System.Object obj, System.Object[] parameters) [0x00014] in <e1319b7195c343e79b385cd3aa43f5dc>:0
  at System.RuntimeType.CreateInstanceMono (System.Boolean nonPublic) [0x000a8] in <e1319b7195c343e79b385cd3aa43f5dc>:0 
  at System.RuntimeType.CreateInstanceSlow (System.Boolean publicOnly, System.Boolean skipCheckThis, System.Boolean fillCache, System.Threading.StackCrawlMark& stackMark) [0x00009] in <e1319b7195c343e79b385cd3aa43f5dc>:0
  at System.RuntimeType.CreateInstanceDefaultCtor (System.Boolean publicOnly, System.Boolean skipCheckThis, System.Boolean fillCache, System.Threading.StackCrawlMark& stackMark) [0x00027] in <e1319b7195c343e79b385cd3aa43f5dc>:0
  at System.Activator.CreateInstance (System.Type type, System.Boolean nonPublic) [0x0002c] in <e1319b7195c343e79b385cd3aa43f5dc>:0
  at System.Activator.CreateInstance (System.Type type) [0x00000] in <e1319b7195c343e79b385cd3aa43f5dc>:0
  at MelonLoader.MelonHandler.LoadFromAssembly (System.Reflection.Assembly asm, System.Boolean isPlugin, System.String filelocation) [0x000bb] in <b506e9dd64bf4ed5b37d9d05c523c205>:0
  at MelonLoader.MelonHandler.LoadFromFile (System.String filelocation, System.Boolean isPlugin) [0x0001a] in <b506e9dd64bf4ed5b37d9d05c523c205>:0
  at MelonLoader.MelonHandler.LoadAll (System.Boolean plugins) [0x0008a] in <b506e9dd64bf4ed5b37d9d05c523c205>:0
[/code]

benaclejames, 2021-01-20:
Thanks for the bug reports! I've released a fix for the issue [here](https://github.com/benaclejames/VRCEyeTracking/releases/tag/v1.2.1). Please note that this is not the final build for this update as I still need to figure out how to prioritize the network traffic which I hope to have done soon. For time time being, however, this pre-release should fix the main issue! Please let me know if you encounter any further issues!

closed 2021-01-20 by benaclejames