benaclejames/VRCFaceTracking #13: Add HTC Face tracker support.

Yersi88, 2021-04-04:
Could you please add support for face tracking with new htc face tracker? It could be straight forward as it uses SRanipal runtime as well.

benaclejames, 2021-04-04:
Work is being done to support this on the [lip-tracking branch](https://github.com/benaclejames/VRCEyeTracking/tree/lip-tracking)! It's not done yet but it works somewhat for now. Currently working on getting the SDK to initialize both eye and lip tracking properly and then I'll release the first iteration. I'm also working on a UI update which will come later 😁
[image]

Yersi88, 2021-04-04:
> Work is being done to support this on the [lip-tracking branch](https://github.com/benaclejames/VRCEyeTracking/tree/lip-tracking)! It's not done yet but it works somewhat for now. Currently working [...]

Oh wow, sorry I didn't notice you're already on it! Can't wait to test it out, thanks for your hard work :)

NamBo77, 2021-04-19:
Any planned time for this to be released? I am very exited to get to try this out! Really appreciate the work!

benaclejames, 2021-04-19:
It's fairly difficult to test stuff without the hardware. I've been guessing this whole time 😆
Right now I'm working on merging some parameters into single negative and positive parameters to save on network bandwidth.

NamBo77, 2021-04-19:
That makes sense. I have the facial tracker myself so I would be happy to help in any way but im not very knowledgeble in coding or anything like that.

AmaliaKalio, 2021-04-20:
Thanks for merging in the lip tracker code to the main branch! I understand it's still a work in progress, but what do we need to do in order to tie it into our avatars? Since I'm not seeing a LipTrackingParams.cs to mirror the EyeTrackingParams.cs file, I'm not sure what we need to do for parameters.

benaclejames, 2021-04-20:
If you're willing to build it yourself, you can uncomment from line 52 to line 61 inside SRanipalTrackingInterface.cs and the experimental lip tracking will initialize. A list of the available parameters can be found [here](https://github.com/benaclejames/VRCEyeTracking/blob/203742753feeecb4ea8a845295ee4a7b5fd4cdae/VRCEyeTracking/SRanipal/Lip/SRanipal_LipData_v2.cs#L11)

AmaliaKalio, 2021-04-20:
Having trouble tracking down all of the dependencies to build it on my machine. I know that's entirely on me, but I'm curious why it's commented out if there's already a check for if the hardware exists? (I could be misinterpreting that.)

Likewise, wouldn't a list of params this long exceed the limit allowed per avatar? Or is this somehow natively hooking into the existing visemes?

benaclejames, 2021-04-20:
It's commented out at the moment as it's somewhat untested. I wouldn't want to release something jank lol
I plan to create a merging system for those parameters as there are way too many to fit on an avatar, hence the unfinished nature of the code.

benaclejames, 2021-04-24:
Please let me know if the pre-release works well! 
https://github.com/benaclejames/VRCEyeTracking/releases/tag/v2.0.0a

NamBo77, 2021-04-24:
Yeah im very much too inexperienced with this stuff to make my own avatar out of this lol. Stopped making avatars for over a year now. Is there any already made avatars that I could atleast test facial tracking with at the moment? Maybe ill have to commision for the feature to be added in an avatar later on or something

Yersi88, 2021-04-24:
> Yeah im very much too inexperienced with this stuff to make my own avatar out of this lol. Stopped making avatars for over a year now. Is there any already made avatars that I could atleast test [...]

Well you have tutorial in the Wiki but it's not that straight forward let me tell you that! But if you dedicate some time you can do it. Also I don't think there are any avatars that are 'pre-made' for face tracking at the moment and I doubt someone will give you theirs avatar that they put much time into setting up :P

NamBo77, 2021-04-24:
Yeah that is true. Im going to hold off on it a little bit but yeah i was thinkin a bit more like a demo but i understand people keeping it to themselves. Can't expect to get stuff for free lul. I'd gladly support someone for making an avatar with this in it tho

InsaneGrox, 2021-04-24:
I might set one up sometime once the facial tracking build is more complete, It's probably going to be either a generic E-boy or a slightly modified version of HTC's sample model.

benaclejames, 2021-04-24:
> I might set one up sometime once the facial tracking build is more complete, It's probably going to be either a generic E-boy or a slightly modified version of HTC's sample model.
> (It will be built for eye and lip tracking so sorry non-vive pro eye users, your eyes will be perma-shut)

I'd include a link to it in the wiki if you beat me to it 😄

allbadbadgers, 2021-04-25:
Sorry this is a small question, I'm a bit new to github. Is there a way I can get email updates about the project from github? or should i just check here periodically?

benaclejames, 2021-04-25:
> Sorry this is a small question, I'm a bit new to github. Is there a way I can get email updates about the project from github? or should i just check here periodically?

Feel free to watch the [commit log](https://github.com/benaclejames/VRCEyeTracking/commits/master). Whenever I push new code, you'll see it show up here. I try my best to describe what I changed and you can see exactly what I changed if you need more details 😄

InsaneGrox, 2021-05-05:
> I'd include a link to it in the wiki if you beat me to it 😄

https://drive.google.com/file/d/1rNnhwWxp3OQSgd1D0jRCWdXaQf4Srmst/view?usp=sharing
It's a modified nikkei because nostalgia... also because it seemed perfect due to the facial shape being very human... (will say it's a bit cursed but maybe people can use it as a study guide or a model to test if the mod is working... I tested it myself so it should work though it does require rero's modified standard shaders)

allbadbadgers, 2021-05-06:
Hey, I saw the model you'd made simply to see if it worked, not knowing if that did work or not I was willing to give it a try. I got the model uploaded in unity as the package it was set up as and everything but the mod is confusing me. for insulation you stated "As is standard, just drag the VRCFaceTracking.dll into your mods folder!". Though when you install the mod it comes in a zip folder consisting of over 4 other folders and only one of them containg 5 different .dll files that all werent just "VRCFaceTraching.dll"

Now too sure what to do from here to try any of this stuff.

NamBo77, 2021-05-06:
> Though when you install the mod it comes in a zip folder consisting of over 4 other folders and only one of them containg 5 different .dll files that all werent just "VRCFaceTraching.dll"
> 
> 
> 
> Now too sure what to do from here to try any of this stuff.

Im guessing you just downloaded the code and not from the release page. Download the dll here https://github.com/benaclejames/VRCFaceTracking/releases/tag/v2.0.1

allbadbadgers, 2021-05-06:
> Im guessing you just downloaded the code and not from the release page. Download the dll here https://github.com/benaclejames/VRCFaceTracking/releases/tag/v2.0.1

thank ya! do you know if this works? using their avatar and this whole set up?

NamBo77, 2021-05-08:
> thank ya! do you know if this works? using their avatar and this whole set up?

I'm personally not very knowledgeble in how to set it up im still trying to figure it out. check out the wiki here for a setup guide
https://github.com/benaclejames/VRCFaceTracking/wiki

closed 2021-05-26 by benaclejames