benaclejames/VRCFaceTracking #15: Can you add support Pimax Droolon Pi1

Kieaer, 2021-04-09:
API+Document can download when install Pimax + Eye tracking module
Not SRanipal. Language is C#. Can you add support it?.

Rexios80, 2021-04-09:
Bump. If you need access to the sdk I can send you the files. It's pretty much impossibly to get without owning a Pimax with the eye tracking module.

benaclejames, 2021-04-09:
Hmm, the mod is pretty much built around SRanipal at the moment but I'd be happy to take a look at the SDK and weigh up how easy/hard it'd be to implement 😁

Kieaer, 2021-04-10:
@benaclejames Here's the sdk documentation.
[aSeeVR Doc.zip](https://github.com/benaclejames/VRCEyeTracking/files/6289091/aSeeVR.Doc.zip)

benaclejames, 2021-04-11:
Thanks! Did the SDK come with any .dll libraries I can use? I can't seem to find one in the provided zip

Rexios80, 2021-04-12:
Here's the whole folder: https://we.tl/t-1coAgawvUN

Kieaer, 2021-04-12:
@benaclejames Here
https://drive.google.com/file/d/1-nf9lCu7rdHKhrQ0SRL192WcQ0htRltT/view?usp=sharing
or use Rexios80's download link if doesn't work google drive link

benaclejames, 2021-04-12:
Oh god it's all in Chinese 😂

NGenesis, 2021-04-13:
From my testing of their (C++, not C# as OP mentioned) SDK, most of the metrics they expose are currently not working. The only ones I was able to get data out of are:

- Openness - seems to be restricted to three states - 0: Fully closed, 50: Half way (very unreliable and causes a lot of fluttering of the eyes) or 100: Fully open)
- Pupil Center X - normalized between 0 (Left) and 1 (Right), seems to stop tracking outside of the 0.25 - 0.75 range.
- Pupil Center Y - normalized between 0 (Up) and 1 (Down), seems to stop tracking outside of the 0.25 - 0.75 range.
- Recommended Eye (seems to always be set to the Right eye from my testing)
- Gaze Origin - not even sure what the representation for this data is.

[image]
[image]

Kieaer, 2021-04-16:
[image]
recommand value can check when calibration finish

benaclejames, 2021-04-22:
Closing due to inactivity. The pimax's eye tracking module seems somewhat sub-par and has an even smaller userbase than the already-niche SRanipal tracking userbase. Supporting it would be something I may work on in the future but I'm more focused on lip tracking at the moment

closed 2021-04-22 by benaclejames

NGenesis, 2021-05-07:
I started working on a mod for supporting the subset of functionality that the Droolon Pi1 provides for now at https://github.com/NGenesis/VRCPimaxEyeTracker

Kieaer, 2021-06-12:
Nice.