benaclejames/VRCFaceTracking #16: Potentially Broken - 4/20/21

TakkinFaircrest, 2021-04-20:
Was using it fine yesterday; however today I'm not getting any Eye Tracking or Blink Detection. Not sure yet if it's just me or others as well.

[02:00:16.766] VRCEyeTracking v1.3.0
[02:00:16.768] by benaclejames
[02:00:16.781] SHA256 Hash: 1b942e30558ab5118613cb230bff6ecb3ce1bfdb5c7f3f9dd32dae04cc672589
[02:00:16.782] ------------------------------
[02:00:16.784] [emmVRCLoader] [emmVRCLoader] Checking if emmVRC can Load...
[02:00:17.406] [emmVRCLoader] [emmVRCLoader] emmVRC can be loaded!
[02:00:17.407] [emmVRCLoader] [emmVRCLoader] Downloading emmVRC...
[02:00:18.338] [emmVRCLoader] [emmVRCLoader] OnApplicationStart called

Steam_SetMinidumpSteamID:  Caching Steam ID: **ManuallyRemoved** [API loaded no]
[02:00:21.321] [emmVRCLoader] VRChat build is: 1078
[02:00:21.321] [emmVRCLoader] Initializing network...
[02:00:21.661] [emmVRCLoader] [emmVRC] Initializing Quickmenu HUD
[02:00:22.631] [emmVRCLoader] Initialization is successful in 01.3s. Welcome to emmVRC!
[02:00:22.632] [emmVRCLoader] You are running version 2.8.5
[02:00:22.638] [VRCEyeTracking] Initializing SRanipal...
[02:00:30.759] [VRCEyeTracking] SRanipal SDK Started. Eye Tracking Active

InsaneGrox, 2021-04-20:
This also affects me, seems like VRC update borked it again...

benaclejames, 2021-04-20:
Fixed! 😄

closed 2021-04-20 by benaclejames