benaclejames/VRCFaceTracking #19: Make eyes not go to bottom right corner when losing tracking

InsaneGrox, 2021-04-21:
Currently when closing your eyes or the eyes lose tracking they revert to the bottom right corner... this would be fine if it wasn't for VRchat's network interpolation smoothing the eye tracking out to the point of causing it to be very noticable... I have two ideas for fixing this
1: make the blend tree parameters revert to 0.5 0.5 instead of 0 0... (this would somewhat fix it but still be a bit jank)
2: make the eyes retain their last look direction when losing tracking (this would probably look the best)

benaclejames, 2021-04-22:
Hmm, solid idea! I could perhaps just not update the "look" params if the corresponding eyelid is closed beyond around 0.05

benaclejames, 2021-04-22:
Nevermind! Turns out SRanipal has their own data valitity checker for the returned tracker info! I can just check if the gaze direction is valid and, if not, just not update the param

benaclejames, 2021-04-24:
Fixed in pre-release. Will be included in next full release.
https://github.com/benaclejames/VRCEyeTracking/releases/tag/v2.0.0a

closed 2021-04-24 by benaclejames