benaclejames/VRCFaceTracking #20: Add combined eyelid parameter (for parameter opimization)

InsaneGrox, 2021-04-21:
Idea: add combined eyelid parameters (left right and combined) that includes widening openness and squeezing...
EG: 0-0.2 = widen... 0.2-0.8=blink... 0.8-1.0 = squeeze...

Reason: To allow for more room for extra parameters with lip tracking.

benaclejames, 2021-04-22:
[image]
Combined parameters already exist, not sure if this is what you're suggesting

InsaneGrox, 2021-04-23:
I was saying combine them even more... from what I've seen you cannot widen without your eyes being fully open... and you cannot squeeze without your eyes being fully closed...
therefore... have a combined parameter that combines them even more (mostly because you can have more than 2 animations on a 2d blend tree)

benaclejames, 2021-04-23:
Ooo! Awesome idea!! I'll get to work on that very soon

benaclejames, 2021-04-23:
Just committed a working version with the merging. Unfortunately, the squeeze data is often inaccurate and I'm yet to find anyone who actually uses it so I left it out of the merge. The new parameters are called "LeftEyeLidExpanded" and "RightEyeLidExpanded". Both scale blinking to 0-0.8 and widening to 0.8-1.

benaclejames, 2021-04-24:
Added in pre-release. Will be in next full release.
https://github.com/benaclejames/VRCEyeTracking/releases/tag/v2.0.0a

closed 2021-04-24 by benaclejames