benaclejames/VRCFaceTracking #21: Mod does not initialize lip tracking without eye tracking

Juspertinry, 2021-04-24:
When running the mod with only lip tracking (Using Valve Index) I recieve the following output/error

[05:10:21.262] [VRCEyeTracking] Initializing SRanipal...
[05:10:26.309] [VRCEyeTracking] [WARNING] Eye Tracking will be unavailable for this session. (NOT_SUPPORT_EYE_TRACKING)
[05:10:26.313] [VRCEyeTracking] SRanipal Lip Initialized!

After this, the UI is never given the configuration button.

benaclejames, 2021-04-24:
The UI is still work-in-progress, hence the button not showing at the moment. As for the warning, I could disable it for eye tracking failures if that'd be any better. I assume most people who use the mod will be using it for the lip tracker anyways

Juspertinry, 2021-04-24:
I think logging missing devices should be fine, just a little confusion on my part as I was just trying everything without a proper avatar first.

closed 2021-04-25 by benaclejames