benaclejames/VRCFaceTracking #22: Pre-release version does not initialize eye/lip tracking with Vive pro eye + Facial tracker

InsaneGrox, 2021-04-24:
Pre-release build does not function at all when I try to use it
latest release build functions flawlessly
Logs
[MelonLoader_21-04-24_14-16-57.886.log](https://github.com/benaclejames/VRCEyeTracking/files/6370335/MelonLoader_21-04-24_14-16-57.886.log)

Video of what happens in-game

https://user-images.githubusercontent.com/54233743/115968978-81be8400-a508-11eb-8adb-8a289771d2a5.mp4

benaclejames, 2021-04-24:
Hmm, looks like it initialized fine but sranipal tried to change values while paramlib was trying to read them. I have replaced the existing dll with a new one that should hopefully fix this issue. Please let me know if the issue persists! 😄

InsaneGrox, 2021-04-24:
Fixed... They initialize properly now...

closed 2021-04-24 by InsaneGrox