benaclejames/VRCFaceTracking #25: SRanipal fails to initialize on latest VRchat version

InsaneGrox, 2021-05-04:
[MelonLoader_21-05-04_00-46-19.627.log](https://github.com/benaclejames/VRCFaceTracking/files/6419042/MelonLoader_21-05-04_00-46-19.627.log)
there's the log, ah shit here we go again...

TakkinFaircrest, 2021-05-04:
Can confirm. Latest VRChat update has broken things.

NGenesis, 2021-05-04:
Looks like it needs to be changed to:

https://github.com/benaclejames/ParamLib/blob/6de64be519cf11b4e5a3bcb03aa914ee294217a5/ParamLib.cs#L12-L13
 ```csharp
       private static AvatarAnimParamController GetLocalAnimParamController() => VRCPlayer.field_Internal_Static_VRCPlayer_0
            ?.field_Private_AnimatorControllerManager_0?.field_Private_AvatarAnimParamController_0;
```

benaclejames, 2021-05-04:
> 

Yup! I have these changes done locally, just trying to figure out where the float param prioritisation went with the update

benaclejames, 2021-05-05:
Fixed in latest release

closed 2021-05-05 by benaclejames