benaclejames/VRCFaceTracking #27: Vive Wireless - Parameters frozen on battery change

lithiumfrog, 2021-05-07:
Noticed that when my Vive wireless battery dies and I swap it out, my parameters are no longer updating. Eye/Face trackers are still working outside of VRC, but to get my parameters to update I'm forced to restart VRC. Tried the force reinitialization button in your menu, but it didn't seem to help. Also tried lesser things like resetting avatar, changing avatars and changing back, nothing seemed to work except restarting VRC completely.

Thank you!

lithiumfrog, 2021-05-12:
Something to note, I found my wireless adapter to be relatively unstable after adding the facial tracker, and discovered my cooling solution was no longer adequate as the facial tracker added nearly 20c of load. I have since added better heatsinks and it solved a lot of this. Is it possible some of the errors you're receiving may be from an overheating adapter? The quality of my facial tracking seems to have improved as well, it was already good, but it's definitely better now.

AmaliaKalio, 2021-05-12:
This is likely related to (and testable from) a similar case I had seen, where if the headset loses power briefly (IE: brief power outage), the same thing happens. Willing to bet you can replicate this on a wired headset by pulling the power cable from the link box.

bizzclaw, 2022-01-23:
Is this closed because it can't be fixed? I have a wireless module and I can confirm the issue. The same thing happens if you pull the power cable and plug it back in. The hard reset button doesn't seem to fix it either.