benaclejames/VRCFaceTracking #33: [Enhancement] Possible OpenSeeFace Integration

dfgHiatus, 2021-05-31:
VRCFaceTracking currently utilizes SRAnipal to drive avatar expressitivity. I would like to propose the creation of a separate branch that would enable Desktop-Mode users to have the same expressitivity via OpenSeeFace.

For reference, me and a friend have created an [application](https://github.com/Ruz-eh/NeosWCFaceTrack) that gives Desktop-Mode users in NeosVR this functionality. Given the overlap our systems have (outputting values to drive parameters), I am curious to see if we could have WCFaceTrack interface with VRC's Avatar Params this way. Please let me know your thoughts! 

https://user-images.githubusercontent.com/51272212/120246630-64cb4e00-c236-11eb-9389-5357eb30d9da.mp4

benaclejames, 2021-06-01:
This seems cool! Though it seems like it requires python to run and the userbase may not be big enough to be worth implementing.
I think before I start adding more frameworks though, I'll need to find a more reliable way to decide which frameworks to use for which features. Perhaps I'll add a requirement for new users to select their preferred runtime in the UI before trying to initialize anything

AmaliaKalio, 2021-06-01:
Not that we have any dates (or concrete details for that matter) on this, but the last VRC dev stream mentioned implementing eye/face tracking from web cam data. While it's unclear if the rest of this mod as it is would be replaced by that, this functionality certainly would be.

Not to say it wouldn't be cool to have, because it really would be.

dfgHiatus, 2021-06-02:
I see your points. However, I have two quick notes:
- What complications would using Python create? WCFaceTrack has a ready executable with Params, users would not need to install Python prior to this
-- The executable sends face/eye information over a websocket connection, all you'd need to do is listen to this instead of SRAnipal
- The majority of VRChat plays on Mouse and Keyboard, IIRC upwards of 70%. We could appeal to a larger player-base this way

> Perhaps I'll add a requirement for new users to select their preferred runtime in the UI before trying to initialize anything

I definitely agree. Still, would this be better suited if this sort of project would have a Repo of its own? After all, the basis for this is VR Eye and Face Tracking, not Desktop-Mode Eye and Face Tracking

closed 2021-07-03 by benaclejames