benaclejames/VRCFaceTracking #40: Pimax eye tracking not working

Kieaer, 2021-06-20:
```
[12:08:27.140] [VRCFaceTracking] Eye Tracking Initialized
[12:08:27.142] [VRCFaceTracking] [WARNING] Lip Tracking will be unavailable for this session.
[12:08:27.155] [ERROR] System.NullReferenceException: Object reference not set to an instance of an object
  at VRCFaceTracking.Pimax.PimaxTrackingInterface.Update (System.Threading.CancellationToken token) [0x00028] in <ba71da3e08bc4e7d927366e0849cad21>:0 
  at VRCFaceTracking.Pimax.PimaxTrackingInterface+<>c.<.cctor>b__11_0 () [0x0000a] in <ba71da3e08bc4e7d927366e0849cad21>:0 
  at System.Threading.ThreadHelper.ThreadStart_Context (System.Object state) [0x00014] in <e1319b7195c343e79b385cd3aa43f5dc>:0 
  at System.Threading.ExecutionContext.RunInternal (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) [0x00071] in <e1319b7195c343e79b385cd3aa43f5dc>:0 
  at System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) [0x00000] in <e1319b7195c343e79b385cd3aa43f5dc>:0 
  at System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state) [0x0002b] in <e1319b7195c343e79b385cd3aa43f5dc>:0 
  at System.Threading.ThreadHelper.ThreadStart () [0x00008] in <e1319b7195c343e79b385cd3aa43f5dc>:0 
  ```

Kieaer, 2021-06-20:
startup crash

retroheim, 2021-09-16:
I second this, same issue.

benaclejames, 2021-11-12:
For all who are still having issues, please test the [latest release](https://github.com/benaclejames/VRCFaceTracking/releases/tag/v2.5.0) which may or may not fix the issues encountered here.

retroheim, 2021-11-13:
[21:31:40.501] [ERROR] System.NullReferenceException: Object reference not set to an instance of an object

  at VRCFaceTracking.Pimax.PimaxTrackingInterface.Initialize (System.Boolean eye, System.Boolean lip) [0x0002a] in <3cadcf3891e64e63a7cf41fda0b37d58>:0 

  at VRCFaceTracking.UnifiedLibManager.FindAndInitRuntimes (System.Boolean eye, System.Boolean lip) [0x0007c] in <3cadcf3891e64e63a7cf41fda0b37d58>:0 

  at VRCFaceTracking.UnifiedLibManager+<>c__DisplayClass6_0.<Initialize>b__0 () [0x00000] in <3cadcf3891e64e63a7cf41fda0b37d58>:0 

  at System.Threading.ThreadHelper.ThreadStart_Context (System.Object state) [0x00014] in <e1319b7195c343e79b385cd3aa43f5dc>:0 

  at System.Threading.ExecutionContext.RunInternal (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) [0x00071] in <e1319b7195c343e79b385cd3aa43f5dc>:0 

  at System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) [0x00000] in <e1319b7195c343e79b385cd3aa43f5dc>:0 

  at System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state) [0x0002b] in <e1319b7195c343e79b385cd3aa43f5dc>:0 

  at System.Threading.ThreadHelper.ThreadStart () [0x00008] in <e1319b7195c343e79b385cd3aa43f5dc>:0

retroheim, 2021-11-13:
It launches, but the moment it start getting into a world it crashes

FynnKitty, 2021-12-09:
still have this problem too

[17:24:38.801] [ERROR] System.NullReferenceException: Object reference not set to an instance of an object

  at VRCFaceTracking.Pimax.PimaxTrackingInterface.Initialize (System.Boolean eye, System.Boolean lip) [0x0002a] in <0c51fed323ec46a5b8505fe6ffb3c0ae>:0 

  at VRCFaceTracking.UnifiedLibManager.FindAndInitRuntimes (System.Boolean eye, System.Boolean lip) [0x0007c] in <0c51fed323ec46a5b8505fe6ffb3c0ae>:0 

  at VRCFaceTracking.UnifiedLibManager+<>c__DisplayClass6_0.<Initialize>b__0 () [0x00000] in <0c51fed323ec46a5b8505fe6ffb3c0ae>:0 

  at System.Threading.ThreadHelper.ThreadStart_Context (System.Object state) [0x00014] in <e1319b7195c343e79b385cd3aa43f5dc>:0 

  at System.Threading.ExecutionContext.RunInternal (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) [0x00071] in <e1319b7195c343e79b385cd3aa43f5dc>:0 

  at System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) [0x00000] in <e1319b7195c343e79b385cd3aa43f5dc>:0 

  at System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state) [0x0002b] in <e1319b7195c343e79b385cd3aa43f5dc>:0 

  at System.Threading.ThreadHelper.ThreadStart () [0x00008] in <e1319b7195c343e79b385cd3aa43f5dc>:0

retroheim, 2021-12-14:
With the latest update, I am able to join a vrchat world, I calibrated and tested my eye tracking outside of it prior to that, and I see a menu, the menu says that I don't have tracking. Just reporting on the changes every update. In the logs it just says eye tracking and lip tracking will be unavailable during this session.

Hopefully this helps in any way possible. Really want to have eye tracking enabled.

closed 2022-04-28 by benaclejames