benaclejames/VRCFaceTracking #42: Lip Tracking Avatars

KnightDemons, 2021-07-12:
Does anyone know of a working Avatar to test of the Lip Tacking? I been trying to get my first avatar edit to work but really want to see if my Facial Tracker even works at this point.

regzo2, 2021-07-15:
I am working on an avatar specifically designed with the mod in mind though I'm not fully finished with it. If you want, I could take a look at your avatar in Unity and get it working for you (or to see that your Facial Tracker actually works)!

GeekAndProud, 2021-07-16:
the sdk already comes with an app to test everything without the need to make an irrelevant issue on this github...

benaclejames, 2021-07-21:
Groxxy made a couple really good avatars using the mod. If you have EMM's avatar search, you might be able to find it by searching things such as "face tracking", "eye tracking" or "lip tracking".
I still need to make some sort of avatar world to allow people to just jump in and try out the avatars.

closed 2021-07-21 by benaclejames

InsaneGrox, 2021-07-23:
The search term I'm going to be using when uploading is "VRCFT" for future reference... I'd advise anyone else making public eye/lip tracking compatible avatars have that in the name...