benaclejames/VRCFaceTracking #44: Serial input from microcontroller/computer devices?

anonmon, 2021-07-20:
Not an issue per se, but it would be super useful if this mod had the ability to read a specifically formatted serial input from a COM port to be able to drive the eye and face parameters.
This would single handedly blow open the door for setting up all sorts of eye and facial input devices by way of setting up and programming a microcontroller / microcomputer to take in face data via all sorts of means and send them along in a relatively easy way to VRChat.
Examples of eye tracking: 
http://onloop.net/hairyplotter/
https://create.arduino.cc/projecthub/H0meMadeGarbage/eye-motion-tracking-using-infrared-sensor-227467
https://www.youtube.com/watch?v=vV4CiNsfUYY / https://arxiv.org/pdf/2003.08499.pdf
Facial tracking would just require any of the relatively straightforward pattern recognition solutions (Literally anything that can run OpenCV or TensorFlow) to be bootstrapped to something that could be mounted to a HMD.

benaclejames, 2021-07-20:
I've had this idea mentioned a couple times now! It would definitely be cool to allow users to create their own forms of eye tracking hardware, though some people may want to run it on the same machine running the game. Maybe a socket server would cover more use-cases. VRC did hint at implementing an OSC layer for avatars in the near future so it may be that the mod would become obsolete before that happens, though I'll definitely be looking into this

anonmon, 2021-07-21:
This official interest in OSC is news to me.
In a possible eventuality that built in OSC functionality for avatars came to VRChat, would this mod transition to being a standalone thing that didn't require loading in as a mod? Considering it seems like a lot of the ground work for integrating into the Vive Pro Eye's (And Pimax's eye tracker?) solution seems to be done already, would seem like a no brainer to make it super easy for anyone to be able to use those eye tracking solutions, regardless if they're a bit squeamish about mods or not. Plus it would be super useful if people rallied around a single 'standard' for implementing eye tracking so it would make it fairly likely if someone implemented eye tracking into a public avatar, and someone has a eye tracking solution, it'd probably "Just work".

closed 2021-08-18 by benaclejames