benaclejames/VRCFaceTracking #48: White blank box in UI tab, not working?

celiaafisher, 2021-08-18:
Hi, I finished setting up my avatar's lip parameters a few weeks ago and set up the mod but when I click on the tab I get the following screen (pic attached) and nothing happens when I move my lips. I am using:

- Valve Index
- VIVE Facial Tracker
- VRChat v. 2021.3.2 (most recent)
- VRChat SDK v 2021.08.11.15.16 (most recent)
- Unity 2019.4.29f1

[image]

200Tigersbloxed, 2021-08-19:
Hi, this feature was removed temporarily removed because it would break SRanipal and the parameters. You can read more about here: #38

celiaafisher, 2021-08-19:
Hi! I was able to figure out what I was doing wrong with your debug menu viewing instructions. I have the following settings for all of my lip tracking parameters (see image) and I feel like the tracking I'm seeing is really inaccurate. Are there better settings I could be using, or is the .1 -> .6 standard? If there are settings you would be willing to share that you're using, I'd greatly appreciate it. Thank you again for your help :) 
[image]

regzo2, 2021-08-19:
I have found that the best way to setup the parameters is to see the values returned in the debug menu, and base your thresholds on those. My values have been as high as 1 with extra room and as low as .7, it just depends on how the Face Tracker interprets your actual face. The .1 is there to be a dead zone so that the animation isn't accidentally activated.

It's also a thing you probably have to play around with too, like for example if you feel that the smile is too much for how much you actually are smiling, then you can increase the threshold of the second value, and vice versa.

celiaafisher, 2021-08-20:
That makes sense! Thanks so much, I'll close the thread.

closed 2021-08-20 by celiaafisher