benaclejames/VRCFaceTracking #52: Face tracking stopped working or is my face tracker broken?

NamBo77, 2021-11-07:
Hi I just booted up vrchat today and everything seemed fine but my avatar wasnt moving with my face movement so i checked the camera preview and it looked wierd, i was wondering if this has been seen before here of if my face tracker is just broken already. the green light on the side goes on and the red light next to the cameras goes on but it just doesnt track my face and the preview is the still image as shown in the image attached below. sorry if this doesnt have anything to do with the mod but i cant find any other way to see if its on my end or not.
[image]

benaclejames, 2021-11-07:
Oof, that looks like a hardware fault! Not sure which part of your hardware has broken, but I'd contact Vive and ask for an RMA

closed 2021-11-07 by NamBo77

NamBo77, 2021-11-07:
> Oof, that looks like a hardware fault! Not sure which part of your hardware has broken, but I'd contact Vive and ask for an RMA

This is really wierd, I tried launching neosvr and it looked fine and then I tried booting up vrchat again and it just works again. I did try restarting my pc and reinstalling the mod but for some reason booting up neos just fixed it. 🤷‍♂️. Sorry for the inconvinience.