benaclejames/VRCFaceTracking #53: VRChat updated!

ShawnMayer, 2021-11-09:
VRChat updated to build 1149.
And borkoen VRFaceTracking mod in now.

[11:58:27.998] [VRCFaceTracking] [ERROR] System.TypeLoadException: Could not resolve type with token 01000078 (from typeref, class/assembly MulticastDelegateNPublicSealedVoGaVRBoUnique, )
  at VRCFaceTracking.MainMod.UiManagerInit () [0x00011] in <1f220596fa2c49ba961a12295f311152>:0
  at VRCFaceTracking.MainMod.CheckUiManager () [0x000ab] in <1f220596fa2c49ba961a12295f311152>:0
  at VRCFaceTracking.MainMod.OnUpdate () [0x00014] in <1f220596fa2c49ba961a12295f311152>:0
  at MelonLoader.MelonHandler+<>c.<OnUpdate>b__48_0 (MelonLoader.MelonBase x) [0x00000] in <67e16b0234254e2183a98652371f8451>:0
  at MelonLoader.MelonHandler+<>c__DisplayClass52_0.<InvokeMelonMethod>b__1 (MelonLoader.MelonMod x) [0x00000] in <67e16b0234254e2183a98652371f8451>:0
  at MelonLoader.MelonHandler.InvokeMelonMethod[T] (System.Collections.Generic.List`1[T]& melons, MelonLoader.MelonHandler+InvokeMelonMethodDelegate`1[T] method, System.Boolean remove_failed) [0x00032] in <67e16b0234254e2183a98652371f8451>:0
[11:58:28.009] [emmVRCLoader] VRChat build is: 1149
[11:58:28.010] [emmVRCLoader] VRChat build is: 1149
[11:58:28.433] [emmVRCLoader] Initialization is successful in 00.4s. Welcome to emmVRC!
[11:58:28.433] [emmVRCLoader] You are running version 3.0.0
[11:58:28.618] [UI_Expansion_Kit] Decorating menus
[11:58:33.078] [VRCFaceTracking] Eye Tracking Initialized
[11:58:33.079] [VRCFaceTracking] Lip Tracking Initialized
TypeRef ResolutionScope not yet handled (139) for .EnumNPublicSealedvaUnCoHeToTaThShPeVoUnique in image data-000001B1E45B5040
[11:58:41.978] [VRCFaceTracking] [ERROR] System.TypeLoadException: Could not resolve type with token 0100008c (from typeref, class/assembly EnumNPublicSealedvaUnCoHeToTaThShPeVoUnique, )
  at VRCFaceTracking.QuickMenu.QuickModeMenu.InitializeMenu () [0x0000f] in <1f220596fa2c49ba961a12295f311152>:0
  at VRCFaceTracking.QuickMenu.QuickModeMenu.CheckIfShouldInit () [0x00007] in <1f220596fa2c49ba961a12295f311152>:0
  at VRCFaceTracking.MainMod.OnSceneWasLoaded (System.Int32 level, System.String levelName) [0x00004] in <1f220596fa2c49ba961a12295f311152>:0
  at MelonLoader.MelonHandler+<>c__DisplayClass44_0.<OnSceneWasLoaded>b__0 (MelonLoader.MelonMod x) [0x0000c] in <67e16b0234254e2183a98652371f8451>:0
  at MelonLoader.MelonHandler.InvokeMelonMethod[T] (System.Collections.Generic.List`1[T]& melons, MelonLoader.MelonHandler+InvokeMelonMethodDelegate`1[T] method, System.Boolean remove_failed) [0x00032] in <67e16b0234254e2183a98652371f8451>:0
TypeRef ResolutionScope not yet handled (139) for .EnumNPublicSealedvaUnCoHeToTaThShPeVoUnique in image data-000001B1E45B5040
[12:02:31.205] [VRCFaceTracking] [ERROR] System.TypeLoadException: Could not resolve type with token 0100008c (from typeref, class/assembly EnumNPublicSealedvaUnCoHeToTaThShPeVoUnique, )
  at VRCFaceTracking.QuickMenu.QuickModeMenu.InitializeMenu () [0x0000f] in <1f220596fa2c49ba961a12295f311152>:0
  at VRCFaceTracking.QuickMenu.QuickModeMenu.CheckIfShouldInit () [0x00007] in <1f220596fa2c49ba961a12295f311152>:0
  at VRCFaceTracking.MainMod.OnSceneWasLoaded (System.Int32 level, System.String levelName) [0x00004] in <1f220596fa2c49ba961a12295f311152>:0
  at MelonLoader.MelonHandler+<>c__DisplayClass44_0.<OnSceneWasLoaded>b__0 (MelonLoader.MelonMod x) [0x0000c] in <67e16b0234254e2183a98652371f8451>:0
  at MelonLoader.MelonHandler.InvokeMelonMethod[T] (System.Collections.Generic.List`1[T]& melons, MelonLoader.MelonHandler+InvokeMelonMethodDelegate`1[T] method, System.Boolean remove_failed) [0x00032] in <67e16b0234254e2183a98652371f8451>:0
TypeRef ResolutionScope not yet handled (139) for .EnumNPublicSealedvaUnCoHeToTaThShPeVoUnique in image data-000001B1E45B5040
[12:03:31.892] [VRCFaceTracking] [ERROR] System.TypeLoadException: Could not resolve type with token 0100008c (from typeref, class/assembly EnumNPublicSealedvaUnCoHeToTaThShPeVoUnique, )
  at VRCFaceTracking.QuickMenu.QuickModeMenu.InitializeMenu () [0x0000f] in <1f220596fa2c49ba961a12295f311152>:0
  at VRCFaceTracking.QuickMenu.QuickModeMenu.CheckIfShouldInit () [0x00007] in <1f220596fa2c49ba961a12295f311152>:0
  at VRCFaceTracking.MainMod.OnSceneWasLoaded (System.Int32 level, System.String levelName) [0x00004] in <1f220596fa2c49ba961a12295f311152>:0
  at MelonLoader.MelonHandler+<>c__DisplayClass44_0.<OnSceneWasLoaded>b__0 (MelonLoader.MelonMod x) [0x0000c] in <67e16b0234254e2183a98652371f8451>:0
  at MelonLoader.MelonHandler.InvokeMelonMethod[T] (System.Collections.Generic.List`1[T]& melons, MelonLoader.MelonHandler+InvokeMelonMethodDelegate`1[T] method, System.Boolean remove_failed) [0x00032] in <67e16b0234254e2183a98652371f8451>:0

HigorMoreno, 2021-11-12:
I'm having the same issue with latest release, seem like @benaclejames is working on a fix. I was able to build his latest code and it's working fine for now. If you need further help contact me on discord Choki#0001 and I can help you.

benaclejames, 2021-11-12:
Hopefully [latest release](https://github.com/benaclejames/VRCFaceTracking/releases/tag/v2.5.0) fixes this!
Thanks

closed 2021-11-12 by benaclejames

HigorMoreno, 2021-11-13:
Thx @benaclejames