benaclejames/VRCFaceTracking #62: Cull parameter list

benaclejames, 2021-12-06:
Go through all parameters and make the decision as to whether to deprecate them. As of now, we have far too many parameters for any sane person to comprehend, so undocument the rarely-used ones and deprecate ones that have newer and more efficient counterparts.

benaclejames, 2021-12-21:
Current proposed list of parameters to deprecate:

> LeftEyeY
> RightEyeY
> LeftEyeLid
> RightEyeLid
> CombinedEyeLid
> EyesWiden
> EyesSqueeze
> LeftEyeWiden
> RightEyeWiden
> LeftEyeSqueeze
> RightEyeSqueeze
> LeftEyeLidExpanded
> RightEyeLidExpanded

> **Left/RightEyeLidExpandedSqueeze will be renamed to Left/RightEyeLid**

All deprecated parameters will continue to work for 3 months.. On avatar load, VRCFT will check for deprecated parameters, reroute them. If the current user is the owner of the loading avatar, they will be shown a warning to remove the deprecated parameters every time they load into it.

InsaneGrox, 2021-12-21:
LeftEyeLidExpanded
RightEyeLidExpanded
Ah shit I guess I'm re-doing every single avatar I have

Adjerry91, 2021-12-22:
This is going cause some serious compatibility issues with LeftEyeLid and RightEyeLid if you are redefining them.

benaclejames, 2021-12-22:
> This is going cause some serious compatibility issues with LeftEyeLid and RightEyeLid if you are redefining them.

The renaming is still TBD, listening to feedback first and so far, it's likely it won't happen. Though Left/RightEyeLid will likely be undocumented/discouraged in the documentation in the future

fuzzy-replex, 2021-12-24:
EyesWiden gone? *slowly catches on fire* "This is fine :)"

benaclejames, 2022-01-24:
Some parameters were removed/changed, but the need to clear up the codebase was removed with the inclusion of EParams. Will close this until further notice!

closed 2022-01-24 by benaclejames