benaclejames/VRCFaceTracking #64: Disable automatic starting and restarting of SRanipal

floriegl, 2021-12-18:
I always get the UAC promo from sr_runtime.exe when I start VRChat in desktop or without my facial tracker plugged in and SRanipal running. When I want to use facial tracking I always check if I have the Runtime running anyway because it's a bit flaky with actually detecting the tracker. I would like to have a toggle in UI Expansion Kit mod settings or even a flag in the MelonPreferences would be enough to disable this behavior, so I can start VRChat without a connected facial tracker or in desktop mode without needing to walk back to my PC and accepting the UAC prompt. Maybe it even makes sense to not start the mod in desktop mode by default in any circumstance because you can't use facial or lip tracking without using an additional dll for facial tracking with a webcam.

200Tigersbloxed, 2021-12-18:
I think since VRCFT doesn't have a config, we could add a launch argument, maybe like `--vrcft-nodesktop` so that people who don't have a desktop tracking device can just disable the mod completely. This feature can be really useful for me especially, as I can't tell you how many times I've launched vrchat and I get a UAC popup; Can be quite irritating from time to time lol.

floriegl, 2021-12-18:
Why not just add a config? I could easily do that, but I am not sure where the runtime gets started

benaclejames, 2021-12-18:
> without my facial tracker plugged in and SRanipal running

There is no way for me to tell if there's a tracker plugged in without first initializing SRanipal. Theoretically I might be able to check connected devices, but that too would require UAC. SRanipal will also not ask you to authenticate UAC a second time if it's already running.

> Why not just add a config

I will not be adding a config for 2 toggles. This seems somewhat unnescessary

I will, however, change the desktop functionality of the mod to be disabled by default in the next update. Then users can re-enable it with a launch parameter.

floriegl, 2021-12-18:
> There is no way for me to tell if there's a tracker plugged in without first initializing SRanipal.
https://github.com/benaclejames/VRCFaceTracking/blob/ffce87939014b92ae9cd6cc6e0512974ac4d677f/VRCFaceTracking/UnifiedLibManager.cs#L44 you could check if the sr_runtime exe is running

200Tigersbloxed, 2021-12-18:
> > There is no way for me to tell if there's a tracker plugged in without first initializing SRanipal.
> 
> 
> 
> https://github.com/benaclejames/VRCFaceTracking/blob/ffce87939014b92ae9cd6cc6e0512974ac4d677f/VRCFaceTracking/UnifiedLibManager.cs#L44 you could check if the sr_runtime exe is running
> 
> 

Some people might not remember to launch the SRRuntime (like me) and have to restart the whole game just to launch the runtime.

floriegl, 2021-12-18:
Don't you have the Recalibrate Button in the QM to (re)start the sr_runtime? Maybe we could have a start-parameter for this too or if you want I can make a PR after you implemented the change for desktop to allow the configuration using MelonPreferences.

200Tigersbloxed, 2021-12-19:
> Don't you have the Recalibrate Button in the QM to (re)start the sr_runtime? Maybe we could have a start-parameter for this too or if you want I can make a PR after you implemented the change for desktop to allow the configuration using MelonPreferences.

I'd prefer to not have to run vrchat as administrator; just a me thing

benaclejames, 2021-12-19:
> you could check if the sr_runtime exe is running

This does not tell me if a facial tracker is plugged in.

floriegl, 2021-12-19:
Yes, it does not. My idea was just to give the user the possibility to disable the automatic starting of the sr_runtime due to the initialization of the SRanipal when the mod initially loads. I am well aware that it is very hard to check for the presence of the facial tracker without it (I assume checking connected USB devices would work?). So the mod only uses SRanipal facial tracking if the user explicitly started it beforehand. From what I have seen some other social VR platforms like Neos do it that way and I would appreciate having the possibility to let the mod work in the same way. Maybe a button which allows the user to tell the mod that he or she started or restarted the Runtime while VRChat is already running would be great to allow for late initialization or fixing of any tracking issues without needing to restart VRChat or needing to start VRChat as admin to allow for recalibration.

Even if you think this is not important for this mod I appreciate the following fix:
> I will, however, change the desktop functionality of the mod to be disabled by default in the next update.

benaclejames, 2021-12-20:
> From what I have seen some other social VR platforms like Neos do it that way

Neos starts SRanipal automatically too... Seems clunky to have to start sranipal every time you want to use it.

floriegl, 2021-12-20:
> Seems clunky to have to start sranipal every time you want to use it.

Yes, needing to run back to the PC in VR to accept the UAC is also clunky. One think that popped into my mind was to allow using a task which already has admin provides on execution. If I remember correctly OVR Toolkit uses this to bypass their UAC prompt. I am not sure how much sense it makes to integrate this into this mod because I don't see any reasonable way on how to remove the task afterwards again. Or maybe just checking if a task exists and then executing it if it exists and informing the users how to set it up if they want in the README. If I get to it, I will create a fork which adds this in the next few days which you could check out.

benaclejames, 2022-03-18:
So this is scuffed, but another way around this issue is just to delete the lip sync notification unity app from the sranipal install dir. That's what causes steamvr to launch

closed 2022-07-30 by benaclejames