benaclejames/VRCFaceTracking #65: avatar compatibility issue.

AeunaSuva, 2021-12-21:
seen it working, love the addon.
my issue is that the tutorial isnt very beginner friendly, and is missing alot of steps.
have you considered making a video tutorial or a discord? itd help people who are just getting started and get some more traction for the plugin..
if not, i highly reccommend you do avatar modding commissions, i garuntee people would pay you to get their avatars setup with this. i know of several people whod like to have their avi modded for eye/face tracking but dont understand how. im sure if you dont do it someone else probably will and better the creator do it right than someone else do it wrong and have people thinking your plugin is at fault/not functional.

Adjerry91, 2021-12-21:
So I did not realize that Facetracking readme does not have dicord link. If you need extra help please join the [Facetracking Discord](https://discord.com/invite/Fh4FNehzKn). On the discord you can find prefabs for eye tracking and face tracking that can be added to avatar controllers. The wiki current covers most of the basics, but video definitely would help more visual learners.

Adjerry91, 2021-12-21:
I haven't looked at wiki in some time. Reviewing it again I think there is room for improvement and adding best practices.

benaclejames, 2021-12-21:
I'll have to add the discord link in README.md soon. The mod is designed for experienced avatar creators and the wiki and it's subsequent guides are designed to be supplimentary to existing knowledge, though I should've made this more clear. I think I'll have to make an commisioner advertise channel in the discord server for people who just want to have someone else set their avatar up with the mod. 😀

AeunaSuva, 2021-12-21:
man i really hope im not stupid and just didnt see it, thanks for letting me know.

closed 2021-12-21 by AeunaSuva