benaclejames/VRCFaceTracking #67: Eye frown parameter

ShawnMayer, 2021-12-28:
I thank always for your work.

I've made the same request before.  (https://github.com/benaclejames/VRCFaceTracking/issues/28)
And I found that Eye_Frown parameter works in v2 version from vive forum.

 https://forum.vive.com/topic/8493-no-way-to-track-eyebrows/

I still think if this parameter is added, it will help to express more emotion.
Could you please add this parameter?

regzo2, 2022-01-17:
Even though Eye_frown is an accessible EyeShape parameter within the SDK, the SRanipal runtime seemingly doesn't return a value for it. Someone would have to add it back into the runtime for it to work. 

I could do a PR to add the parameter in-case anyone adds it to the runtime somehow, or if HTC does 😂.

benaclejames, 2022-01-24:
I think it's safe to say SRanipal is pretty much abandonware at this point. I seriously doubt HTC would be adding it anytime soon. If/when they do, we can re-open this issue and include it as a parameter 😁

closed 2022-01-24 by benaclejames