benaclejames/VRCFaceTracking #68: Eye tracking not working,,, but working?

RaliiTheAvali, 2022-01-02:
So not sure whats going on and i have tried to debug my situation but nothing seems to work. I have a Eye pro and i have had avatars that in the past have worked with the mod. but for some reason it no longer works... but it does. Please read before asking if i have done something.

1: The software used for the Eye pro is RUNNING.
2: Yes i have restarted my PC.
3: Yes i have uninstalled and reinstalled the mod.
4: Yes i have gone in to make sure that parameters were updated (incase there was a change), and that they are still properly named and have capitalizations where needed.

I know the mod itself is working as because in the quick menu where you can check the status, it clearly shows it knows where my eyes are looking. Just the avatars eyes do not move, and my eye lids do not blink. ( AGAIN my avatars worked in the past but for some reason do not work now)

NamBo77, 2022-01-02:
Having the same thing happen yesterday but with face tracking, the only thing i noticed that was different atleast in my case, was that the preview window is dark/grayed out but i can still see my face moving in the camera preview, so seems like you arent alone on this issue.

holadivinus, 2022-01-02:
not a real solution but should help nambo77: make sure you are using the latest off of github, not the modmanager
Then check out the avatar debug menu to see if the param values are actually changing on the avatar itself
just more info

RaliiTheAvali, 2022-01-02:
> Having the same thing happen yesterday but with face tracking, the only thing i noticed that was different atleast in my case, was that the preview window is dark/grayed out but i can still see my face moving in the camera preview, so seems like you arent alone on this issue.

i noticed that the menu was rather grey as well

NamBo77, 2022-01-02:
> not a real solution but should help nambo77: make sure you are using the latest off of github, not the modmanager Then check out the avatar debug menu to see if the param values are actually changing on the avatar itself just more info

What avatar debug menu are you taling about exactly?

RaliiTheAvali, 2022-01-02:
> not a real solution but should help nambo77: make sure you are using the latest off of github, not the modmanager Then check out the avatar debug menu to see if the param values are actually changing on the avatar itself just more info

also i would love to try this,,, but the github location seems to be missing the file,,, i can't find the .dll file

NamBo77, 2022-01-02:
> also i would love to try this,,, but the github location seems to be missing the file,,, i can't find the .dll file

https://github.com/benaclejames/VRCFaceTracking/releases

200Tigersbloxed, 2022-01-02:
> Yes i have gone in to make sure that parameters were updated (incase there was a change), and that they are still properly named and have capitalizations where needed.

Have you verified that the parameters are being updated via. the avatar debug menu, or is your issue that the parameters aren't being updated?

If the parameters are being updated accordingly:
This is an avatar issue, not a mod issue. I'd recommend joining the [discord](https://discord.gg/Fh4FNehzKn) and asking for assistance there.

If the parameters are not being updated:
Verify you have version 2.5.1, and you're on the latest VRChat Build 1160+ (maybe verify game files integrity too?),

To add onto the post above, here's the download link for version 2.5.1 (latest at the time of writing this)
https://github.com/benaclejames/VRCFaceTracking/releases/download/v2.5.1/VRCFaceTracking.dll

RaliiTheAvali, 2022-01-02:
> Have you verified that the parameters are being updated via. the avatar debug menu, or is your issue that the parameters aren't being updated?
>
> If the parameters are being updated accordingly: This [...]

so updating the mod fixed the old avatars but oddly enough a new one i tried to make doesn't so i will join the discord. thanks for the help up to this point :)

closed 2022-01-06 by benaclejames