benaclejames/VRCFaceTracking #75: Support the HP G2 Reverb Omnicept

imagitama, 2022-02-06:
# Idea

Support the HP G2 Reverb Omnicept VR headset's eye and lip tracking.

# Notes

Neos supports the eye tracking (https://github.com/Neos-Metaverse/NeosPublic/issues/2275) but not the lip sync as the API only exposes the raw camera capture not any data (or maybe it does but it isn't well documented).

Lewisvz, 2022-03-04:
I've reached out to HP for more information on the lip tracker.  I'm currently waiting for a response.

Tazzy4311, 2022-04-21:
> I've reached out to HP for more information on the lip tracker. I'm currently waiting for a response.

if you can, I would love to know when you get updated on this'

Lewisvz, 2022-04-21:
HP failed to respond it seems.  I might shoot them another email though.

Tazzy4311, 2022-04-22:
> HP failed to respond it seems. I might shoot them another email though.

For sure, I'll keep you updated as well if I see anything pop up about it

Tazzy4311, 2022-04-28:
> HP failed to respond it seems. I might shoot them another email though.

Nope haven't found anything on it, It seems I might have to change to a Vive pro Eye

MilchZocker, 2022-05-18:
Im a bit late on that but i received a response from HP a while ago(3-4 months ago), but thats where the good news ends sadly:

[image]

benaclejames, 2022-07-30:
A link to @200Tigersbloxed's external tracking module has now been added to the readme. This will likely be the go-to module for omnicept tracking in the future.

closed 2022-07-30 by benaclejames

SummerSigh, 2022-09-23:
https://www.youtube.com/watch?v=KQB8f4X2v6g&t=2s

Time to get the G2 some mouth-tracking baby!

Lewisvz, 2022-09-23:
It's about time.  Hp really let us down.