benaclejames/VRCFaceTracking #82: TongueSteps ends up going from -1 to 1.6, with 1.2-1.6 values happening whenever my lips are out

feilen, 2022-03-30:
[image]

I have TongueSteps setup on my avatar, to where -1 is tongue_in, 0 is tongue_step_1, and 1 is tongue_step_2. I noticed when my lips are out (in a 'pout' expression) my tongue would stick out all the way as well, and noticed the value seems to be getting set higher than should be possible.

VRCFT Version: 3.0.1
SRanipal Version: 1.3.1.1

feilen, 2022-06-21:
This seems to occur less often with a new face tracker mount that puts it at a lower position. Still, quite odd that it even can go up to 1.6, since the value is supposed to be -1 to 1. I should explicitly see what the values of Tongue_Step_1 and Tongue_Step_2 are when it happens.

benaclejames, 2023-03-26:
@regzo2 any ideas on this?

regzo2, 2023-03-29:
Looking at `3.0.1` tagged version of the repo, I can't see where this would happen. TongueSteps converts Step1 and Step2 using this function, with `_steps` == true. From the calculations it shouldn't be from the VRCFaceTracking

https://github.com/benaclejames/VRCFaceTracking/blob/46e88865c0d92f8f15c6a6628226adb5981be1bd/VRCFaceTracking/Params/Lip/LipShapeMerger.cs#L125

`_steps` == true.
https://github.com/benaclejames/VRCFaceTracking/blob/46e88865c0d92f8f15c6a6628226adb5981be1bd/VRCFaceTracking/Params/Lip/LipShapeConversion.cs#L25-L30

Plugging in any number should pass through this formula `(Step2 - (-Step1)) - 1`. Here is how the values should work:

* Plugging in 1 for both should give `(1 - (-1)) - 1` == `1`
* Plugging in 1 for only Step1 (tongue at teeth) should give `(0 - (-1)) - 1` == `0`
* Plugging in 0 for both should give `(0 - (-0)) - 1` == `-1`

So this would really leave it to be from the SRanipal interface, or something in-between the interface and GetBlendedShape.

benaclejames, 2023-06-27:
Stale

closed 2023-06-27 by benaclejames