benaclejames/VRCFaceTracking #107: [Feature Request] Add config API for module developers

dfgHiatus, 2022-11-03:
As part of the Quest Pro module's development, we've required users to specify custom data via a text file - this is an unintuitive approach, and I imagine other module developers have run into a similar situation. I'm basing this feature request given my experience with [NeosModLoader's config system](https://github.com/neos-modding-group/NeosModLoader/blob/master/doc/config.md).

Given a module, a developer could mark any field they wish to expose to VRCFT with an attribute, say `VRCFTConfigurable`. When it comes time to load a module, all VRCFT has to do is examine the module's assembly for fields marked with said attribute [with something as follows](https://stackoverflow.com/questions/15404634/how-to-get-a-list-of-all-fields-with-a-specified-attribute#15453764):

```
var fields = root.DescendantNodes()
                 .OfType<AttributeSyntax>()
                 .Where(a => a.Name.ToString() == "VRCFTConfigurable")
                 .Select(a => a.Parent.Parent)
                 .Cast<FieldDeclarationSyntax>();
```

Once these fields have been loaded, VRCFT builds a UI as a new tab for said module and generates input boxes for each field. As many modules can be loaded at once, it makes sense to have every module generate its own tab. Here's what I have in mind:

[image]

I'm not sure how to enforce type safety here per se, but I'm opening it up to discussion.

regzo2, 2022-11-19:
I would like to build out some kind of syntax for modules to interface into the User Interface with. Currently a lot of changes are being done to some of the internal systems of VRCFaceTracking, so this will be a pretty good opportunity to include this feature.

regzo2, 2023-02-06:
It appears we will be looking into adding the ability to configurate module pages as part of a new UI overhaul, though we have yet to plot out exactly how we will expose it to developers.