benaclejames/VRCFaceTracking #113: Lags out Vrchat, Does not crash app or Vr

Whittz, 2022-12-07:
Unfortunately I have no logs that show this.
As of right now I can recreate the issues just by using the facetracking app and joining a world for about 10 minutes. Not sure if vrchat's OSC pipeline changed?  Not hardware issue. using 12900k and 3090. face tracker camera works with all other Vr games like helios/chilloutvr. Used this app flawlessly until a month ago.

Does not crash VR; I close vrc and the facetracking app, open it all up again and its fixed for another 10 minutes.

benaclejames, 2022-12-07:
Are you using the recommended SRanipal version 1.3.1.1? This is most likely an issue with Vive's software pipeline.

Whittz, 2022-12-07:
it is not. I've used the 1.3.1.1 and the updated one.  Both causes lag. seems like I have no errors or problems through vive's software

benaclejames, 2022-12-09:
The app is open source, please feel free to throw it in a debugger and prove me wrong. It's vive's sdk. 
Thanks

dfgHiatus, 2022-12-15:
Does an app called `lipsyncnotification.exe` take up a lot of your GPU in the background?

Whittz, 2022-12-15:
unfortunately I had little time and little knowledge of debugging it.  Switched over to an alternative did seem to fix the issue. I'll re upload the avitar for this app and check app usage. but there were times I barely saw gpu/cpu spikes when it happened, and sometimes it was spiking until I closed vrchat

benaclejames, 2023-01-21:
Closing stale thread. Please feel free to re-open once you have the results back between the two apps

closed 2023-01-21 by benaclejames