benaclejames/VRCFaceTracking #114: High GPU Usage

SaladOwO, 2022-12-15:
When launching the face tracking program something called "lipsyncnotification.exe" takes up 98% of my gpu, which is a RTX 3080. Closing it has no affect on the program. What is the reason for this? I suspected a miner.

dfgHiatus, 2022-12-15:
`lipsyncnotification.exe` is part of SRanipal, the eye and face tracking software VRCFaceTracking uses. It is known for not closing down properly once it is launched, you can just delete or rename the executable in your install folder. You can find it relative to SRanipal like in this photo 

[image]

SaladOwO, 2022-12-15:
I see. Thank you 👍

closed 2023-01-04 by SaladOwO