benaclejames/VRCFaceTracking #115: Pimax eye tracking support is gone?

Kieaer, 2022-12-21:
Looking at PimaxEyeTracking in the README, this is a MelonLoader only mod.

Kieaer, 2022-12-21:
```
[12:07:42] VRCFT Standalone Initializing!
[12:07:42] Loaded library C:\Users\cloud\AppData\Roaming\VRCFaceTracking\StockLibs\libHTC_License.dll
[12:07:42] Loaded library C:\Users\cloud\AppData\Roaming\VRCFaceTracking\StockLibs\nanomsg.dll
[12:07:42] Loaded library C:\Users\cloud\AppData\Roaming\VRCFaceTracking\StockLibs\SRWorks_Log.dll
[12:07:42] Loaded library C:\Users\cloud\AppData\Roaming\VRCFaceTracking\StockLibs\ViveSR_Client.dll
[12:07:42] Loaded library C:\Users\cloud\AppData\Roaming\VRCFaceTracking\StockLibs\SRanipal.dll
[12:07:42] Initialized DependencyManager Successfully
[12:07:42] Initialized UnifiedLibManager Successfully
[12:07:42] Finding and initializing runtimes...
[12:07:42] Loading External Modules...

Unhandled Exception: System.ArgumentException: Directory in URL is invalid.
   위치: System.Security.Util.DirectoryString.CreateSeparatedString(String directory)
   위치: System.Security.Util.URLString.ParseFileURL(String url)
   위치: System.Security.Util.URLString.GetFileName()
   위치: System.Reflection.RuntimeAssembly.InternalLoadAssemblyName(AssemblyName assemblyRef, Evidence assemblySecurity, RuntimeAssembly reqAssembly, StackCrawlMark& stackMark, IntPtr pPrivHostBinder, Boolean throwOnFileNotFound, Boolean forIntrospection, Boolean suppressSecurityChecks)
   위치: System.Reflection.RuntimeAssembly.InternalLoadFrom(String assemblyFile, Evidence securityEvidence, Byte[] hashValue, AssemblyHashAlgorithm hashAlgorithm, Boolean forIntrospection, Boolean suppressSecurityChecks, StackCrawlMark& stackMark)
   위치: System.Reflection.Assembly.LoadFrom(String assemblyFile)
   위치: VRCFaceTracking.UnifiedLibManager.LoadExternalModules()
   위치: VRCFaceTracking.UnifiedLibManager.FindAndInitRuntimes(Boolean eye, Boolean lip)
   위치: System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   위치: System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   위치: System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
   위치: System.Threading.ThreadHelper.ThreadStart()
```

regzo2, 2022-12-21:
https://github.com/benaclejames/VRCFTPimaxModule

Kieaer, 2022-12-21:
@regzo2 Load failed because same error as above

Kieaer, 2022-12-21:
[image]

Parameter not changed.

benaclejames, 2023-02-27:
VRCFT cannot provide support for external modules. Please contact the creator of the external tracking module that you've been having trouble with for support.

closed 2023-02-27 by benaclejames