benaclejames/VRCFaceTracking #117: Request/Issue: Vive Focus 3 Eye tracking not working properly

TKDoggo, 2022-12-24:
While using the eye tracker in vrchat, eyelid movement is not present and eye expressions don't work properly.

Easz-eachan, 2023-03-19:
I have same problems. I tried at first, updated some .dll files from v1.3.2.0 to v1.3.6.8 and confirmed to show both eye lid parameter of OSC debug on VRChat. However, these paremeters movement was not good changing between 0.9-1.0 only. I will continue to challenge become to work as well. But, I am confused because I am not sure how to detect the eye lid and I do not know if my tracker is broken or what part of the software is not picking it up properly. I am in need of a lot of info!!
Please share your notice if you guys know or update about that. Please help me, thank you!

regzo2, 2023-03-20:
This is something that is already addressed and fixed in #122, the next VRCFT release will have the newer SRanipal libraries included which will allow the Focus 3 to work. We are currently working out how to keep backwards compatibility with previous SRanipal runtimes as the newer SRanipal libraries cause crashes on older SRanipal runtimes (pre-1.3.6.X) unfortunately.

closed 2023-03-25 by benaclejames