benaclejames/VRCFaceTracking #127: SRanipalExtTrackingInterface Not Detected as Running

GilbertRPG, 2023-04-02:
Initial launch of app shows eye tracking as functional, but states `"An error occurred while getting eye data. This might be a wireless crash."`

I've attempted multiple restarts, re-launches, changes in compatibility, uninstalls. Not sure what's causing this.

Running Windows 11.

regzo2, 2023-04-02:
More context would be appreciated. How you have the hardware configured for SRanipal? Are you using a Vive Wireless adapter, or using wired? 

There are a lot of things that can happen that can cause this error but they usually stem from the hardware side of your setup. This can usually be due to wireless error handling or from bandwidth issues on the hardware (bad cables being a main contributor).

GilbertRPG, 2023-04-02:
Wired Vive Pro Eye.

You mention cable issues. I'm wondering if I have that going for me. Visual display is fine but I was experiencing very jarring loss of tracking/grey view screens, and the occasional black screen, but no actual windows disconnect sound. 

I'll place an order for a replacement cable in the meantime.

GilbertRPG, 2023-04-04:
Turns out I'm a dumbass and forgot to do the initial eye calibration. When you first install drivers for VPE, it will ask for a disclaimer regarding eye tracking. If you don't do this, SRanipal fails to fully initialize and causes VRCFT to think that the eye data hasn't synced up. Awesome to finally figure that one out, I appreciate the timely response.

closed 2023-04-04 by GilbertRPG