benaclejames/VRCFaceTracking #135: Add to winget-pkgs for easy installation and upgrades

rrazgriz, 2023-04-24:
[`winget`](https://learn.microsoft.com/en-us/windows/package-manager/winget/) is a Microsoft-developed open-source package manager for Windows, and its official repo ([`winget-pkgs`](https://github.com/microsoft/winget-pkgs)) readily accepts PRs for new listings. It's fairly easy to author a manifest using [`winget-create` or `YamlCreate.ps1`](https://github.com/microsoft/winget-pkgs#authoring-a-manifest). Updating listings is also done through PRs, and can even be automated with CI.

With VRCFT's latest release coming with nice msix/appinstaller format, it'd be an easy addition to the repository, and more packages helps support a robust package management ecosystem on the Windows platform. It'd make installation as simple as running `winget install BenacleJames.VRCFaceTracking`

benaclejames, 2023-04-27:
[In progress](https://github.com/microsoft/winget-pkgs/pull/103687)

benaclejames, 2023-04-27:
Done!

closed 2023-04-27 by benaclejames