benaclejames/VRCFaceTracking #139: Focus 3 eyetracking not solved yet

Easz-eachan, 2023-04-27:
I tried the new version and still could not get the eyelids to work with the Focus 3 eye-tracking. Mitchell said next VRCFT release will have the newer SRanipal libraries included which will allow the Focus 3 to work. However, when I looked at the source, I found the same old version as before. Could you please change to the new version?

benaclejames, 2023-04-27:
Should be fixed on latest sranipaltrackingmodule release. If you're on latest vrcft version, it'll auto update the module on launch. Please let me know if it's fixed now 😄

Easz-eachan, 2023-04-28:
I saw the comment of this module page, lol. I verified this problem had solved. Thank you sooo much! That's great software! 😄

closed 2023-04-28 by benaclejames