benaclejames/VRCFaceTracking #143: [Bug] Launching VRCFaceTracking without internet crashes the application immediately

dfgHiatus, 2023-05-08:
As the title states, launching VRCFaceTracking without an internet connection crashes the application. This is because VRCFT attempts to poll installable tracking modules and if it fails to do so it hangs. I imagine this is the case in other places where VRCFT polls the web as well.

# Reproduction
1) Disconnect your PC from the internet (Wifi, Ethernet, etc.)
1) Launch VRCFaceTracking
1) Observe the application closes by itself without warning

## Specs:
- VRCFaceTracking v5.0.2.0
- Windows 11 Pro 22H2 22621.1690

See the attached image in `ModuleDataService.cs`:

[image]

benaclejames, 2023-05-08:
Yeeeeah, I realized this would be an issue before release and planned to have a "No Internet Connection" screen display when no connection could be established. I'll work on getting at least the crash fixed but this isn't currently a priority as pretty much everyone who uses VRCFT uses it when playing VRChat, which also requires an internet connection to play.
Thanks for reminding me :P

TheBug233, 2023-05-21:
Not being able to access this address in parts of mainland China will also cause crashes, even if they can play VRChat normally, they have to bypass GFW through VPN first to start VRCFT

closed 2023-06-13 by dfgHiatus