benaclejames/VRCFaceTracking #150: Not Seeing QuestProOpenXRTrackingModule

wjnesham, 2023-06-03:
I'm running Windows 11.

I do see "Quest Pro OpenXR" in modules though.
When I install it with Air Link open, it just displays like a missing image and says Loading...
I see my Quest Pro on the home tab with an "Outgoing" rate, but the "Incoming" rate is always 0 m/s.
Avatar Status is always just "Loading..." even with VRC open through Steam VR.

Log after launching VRCFaceTracking:
[VRCFaceTracking.Services.ActivationService] Information: VRCFT Version 5.0.2.0 initializing...
[VRCFaceTracking.Services.ActivationService] Information: Initializing OSC...
[VRCFaceTracking.Services.ActivationService] Information: Initializing main service...
[VRCFaceTracking.Services.ActivationService] Information: Checking for updates for installed modules...
[VRCFaceTracking.Services.ActivationService] Information: Initializing modules...
[UnifiedLibManager] Information: Starting initialization tracking
[UnifiedLibManager] Information: Tearing down all modules...
[UnifiedLibManager] Information: Initializing runtimes...
[UnifiedLibManager] Information: Initializing VRCFT - Quest OpenXR, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null
[UnifiedLibManager] Information: Loading External Module VRCFT - Quest OpenXR, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null
[QuestProTrackingModule] Information: [QuestOpenXR] OpenXR Runtime init success.
[UnifiedLibManager] Information: Tracking initialized via VRCFT_Module___QuestOpenXR.QuestProTrackingModule

kusomaigo, 2023-06-03:
The module initialized correctly and you're getting face tracking data. the Incoming rate is OSC messages (from VRChat). There is some OSC networking related problem with the game, there appears to be nothing wrong with the module. Next time raise the issue in the Discord first.

benaclejames, 2023-06-27:
Stale

closed 2023-06-27 by benaclejames