benaclejames/VRCFaceTracking #151: Face tracking not working

VaalkinTheOnly, 2023-06-05:
Updated to the newest version and I'm unable to use my lip tracker. Using a Vive facial tracker with SRanipal 1.3.1.1

benaclejames, 2023-06-05:
This is an isolated issue and you've provided little context. I'd recommend posting your logs or asking for help in the [discord](https://discord.gg/vrcft)

VaalkinTheOnly, 2023-06-05:
Logs as follows:
[VRCFaceTracking.Services.ActivationService] Information: VRCFT Version 5.0.2.0 initializing...
[VRCFaceTracking.Services.ActivationService] Information: Initializing OSC...
[VRCFaceTracking.Services.ActivationService] Information: Initializing main service...
[VRCFaceTracking.Services.ActivationService] Information: Checking for updates for installed modules...
[VRCFaceTracking.Services.ActivationService] Information: Initializing modules...
[UnifiedLibManager] Information: Starting initialization tracking
[UnifiedLibManager] Information: Tearing down all modules...
[UnifiedLibManager] Information: Initializing runtimes...
[UnifiedLibManager] Information: Initializing SRanipalExtTrackingModule, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null
[UnifiedLibManager] Information: Loading External Module SRanipalExtTrackingModule, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null
[SRanipalExtTrackingInterface] Information: SRanipalExtTrackingModule: SRanipal version: 1.3.1.1
[UnifiedLibManager] Information: Initializing VRCFTVarjoModule, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null
[UnifiedLibManager] Information: Loading External Module VRCFTVarjoModule, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null
[VarjoTrackingModule] Information: Loaded library C:\Users\never\AppData\Roaming\VRCFaceTracking\CustomLibs\d1bf4744-80c4-4cba-849e-22940511901e\TrackingLibs\VarjoLib.dll
[UnifiedLibManager] Information: Tracking initialized via VRCFTVarjoModule.VarjoTrackingModule

benaclejames, 2023-07-05:
Is this resolved?

UwUo-oUwU, 2023-07-13:
The tracking module for that is now on 1.4, have you tried since then?

Wolfy233, 2023-08-06:
> The tracking module for that is now on 1.4, have you tried since then?

what do yo mean from 1.4

benaclejames, 2023-12-24:
Is this resolved?

spyrohea, 2024-01-11:
Tried that in VRchat and debug console only displays eye tracking. Device is Pico 4 pro.
Strange thing - when I am using wireless connection via Streaming Assistant and opening OSC debug I see face tracking data, but nothing works in VRchat.

closed 2024-04-16 by benaclejames