benaclejames/VRCFaceTracking #152: Reset and Fresh Install both cause app to cease functioning

AmaliaKalio, 2023-06-13:
After the 5.0.2.1 update pushed today, VRCFT was crashing as soon as it loaded parameters. I first attempted a reset from settings, which prevented it from launching. A full uninstall/reinstall gave the same results. Event viewer logs are below, with the referenced .wer attached.
```
Faulting application name: VRCFaceTracking.exe, version: 5.0.2.1, time stamp: 0x641de100
Faulting module name: Microsoft.ui.xaml.dll, version: 3.0.0.2301, time stamp: 0x655f8cbf
Exception code: 0xc000027b
Fault offset: 0x0000000000341e5a
Faulting process id: 0x0xCCFC
Faulting application start time: 0x0x1D99E41F998334F
Faulting application path: C:\Program Files\WindowsApps\96ba052f-0948-44d8-86c4-a0212e4ae047_5.0.2.1_x64__d7rcq4vxghz0r\VRCFaceTracking.exe
Faulting module path: C:\Program Files\WindowsApps\96ba052f-0948-44d8-86c4-a0212e4ae047_5.0.2.1_x64__d7rcq4vxghz0r\Microsoft.ui.xaml.dll
Report Id: b6fd134d-00d9-4d7c-8188-b851d673a657
Faulting package full name: 96ba052f-0948-44d8-86c4-a0212e4ae047_5.0.2.1_x64__d7rcq4vxghz0r
Faulting package-relative application ID: App`
[Report.wer.txt](https://github.com/benaclejames/VRCFaceTracking/files/11739839/Report.wer.txt)
```

```
Fault bucket 1620418211990852622, type 5
Event Name: MoAppCrash
Response: Not available
Cab Id: 0

Problem signature:
P1: 96ba052f-0948-44d8-86c4-a0212e4ae047_5.0.2.1_x64__d7rcq4vxghz0r
P2: praid:App
P3: 5.0.2.1
P4: 641de100
P5: combase.dll
P6: 10.0.22621.1485
P7: a8be94cc
P8: 80070020
P9: 000000000009200b
P10: 

Attached files:
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER.1202249a-a8fd-474b-8c5f-2a0db2ae3764.tmp.mdmp
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER.4d4223fe-48d9-46d3-adfa-47f91f7bedee.tmp.WERInternalMetadata.xml
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER.eaf675ad-259d-4d61-86b2-9d9b266066cf.tmp.csv
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER.7ac0065a-a7ea-4417-91f3-0bb4afd90ce6.tmp.txt
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER.490bfa2f-3642-42c4-a3ce-a7da9080e9f1.tmp.xml

These files may be available here:
\\?\C:\ProgramData\Microsoft\Windows\WER\ReportArchive\AppCrash_96ba052f-0948-44_b3337ded5445080060e37e3e36877beb83fe_a7d66a80_bac571f1-7d0d-4f53-b3fe-0cb58e351757

Analysis symbol: 
Rechecking for solution: 0
Report Id: b6fd134d-00d9-4d7c-8188-b851d673a657
Report Status: 268435456
Hashed bucket: a979f5a916c27145c67ce1c74949d40e
Cab Guid: 0
```

AmaliaKalio, 2023-06-13:
[Report.wer.txt](https://github.com/benaclejames/VRCFaceTracking/files/11739853/Report.wer.txt)
Reattaching wer, as it got removed on post.

AmaliaKalio, 2023-06-13:
Managed to track down an Event Viewer log from when it was crashing before I reset.

```
Application: VRCFaceTracking.exe
CoreCLR Version: 7.0.523.17405
.NET Version: 7.0.5
Description: The process was terminated due to an unhandled exception.
Exception Info: System.NullReferenceException: Object reference not set to an instance of an object.
   at VRCFaceTracking.UnifiedTrackingMutator.Calibrate(UnifiedTrackingData& inputData, Single calibrationWeight)
   at VRCFaceTracking.UnifiedTrackingMutator.MutateData(UnifiedTrackingData input)
   at VRCFaceTracking.UnifiedTracking.UpdateData()
   at VRCFaceTracking.MainStandalone.<InitializeAsync>b__15_1(Object ct)
   at System.Threading.ThreadPoolWorkQueue.Dispatch()
   at System.Threading.PortableThreadPool.WorkerThread.WorkerThreadStart()
```

AmaliaKalio, 2023-06-14:
The program was able to recognize that an update to 5.0.2.2 was available, and launched the updater. After the update was complete, the error scenario in the first post seemed to persist.

benaclejames, 2023-06-14:
Did you try a complete uninstall and reinstall of 5.0.2.2?

AmaliaKalio, 2023-06-15:
> Did you try a complete uninstall and reinstall of 5.0.2.2?

I did, yes. I attempted with both the .msix and .appinstaller files.

AmaliaKalio, 2023-06-16:
This issue persists in 5.0.2.3.

AmaliaKalio, 2023-06-16:
I was able to get v5.0.0.0 to run from a clean install. However, the same crash happened when I attempted to load the 'Module Registry'.

From there, I performed an in-place upgrade to 5.0.1.0. This came with the SRanipal module. It loaded fine and I was also able to access the 'Module Registry' tab.

Attempting an in-place upgrade from there to 5.0.2.3 presented the same immediate fault. Clean-installing back to 5.0.1.0 still works.

Rinse-and-repeat, going down one version at a time. 5.0.1.1 appears to be the latest stable version.

benaclejames, 2023-06-17:
> This issue persists in 5.0.2.3.

Could you provide a latest.log from this release please?

AmaliaKalio, 2023-06-18:
> Owner

```
Faulting application name: VRCFaceTracking.exe, version: 5.0.2.3, time stamp: 0x641de100
Faulting module name: Microsoft.ui.xaml.dll, version: 3.0.0.2301, time stamp: 0x655f8cbf
Exception code: 0xc000027b
Fault offset: 0x0000000000341e5a
Faulting process id: 0x0xEA60
Faulting application start time: 0x0x1D9A09CF01653B2
Faulting application path: C:\Program Files\WindowsApps\96ba052f-0948-44d8-86c4-a0212e4ae047_5.0.2.3_x64__d7rcq4vxghz0r\VRCFaceTracking.exe
Faulting module path: C:\Program Files\WindowsApps\96ba052f-0948-44d8-86c4-a0212e4ae047_5.0.2.3_x64__d7rcq4vxghz0r\Microsoft.ui.xaml.dll
Report Id: 2dab20ca-62b8-459e-a36d-9ccd3d6a2236
Faulting package full name: 96ba052f-0948-44d8-86c4-a0212e4ae047_5.0.2.3_x64__d7rcq4vxghz0r
Faulting package-relative application ID: App
```
```
Fault bucket 1736990634012953323, type 5
Event Name: MoAppCrash
Response: Not available
Cab Id: 0

Problem signature:
P1: 96ba052f-0948-44d8-86c4-a0212e4ae047_5.0.2.3_x64__d7rcq4vxghz0r
P2: praid:App
P3: 5.0.2.3
P4: 641de100
P5: combase.dll
P6: 10.0.22621.1485
P7: a8be94cc
P8: 80070020
P9: 000000000009200b
P10: 

Attached files:
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER.10c036b6-71e4-4ed2-aea2-1299c86c4778.tmp.mdmp
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER.9e5bf8a2-0596-46aa-b847-f366e5381a32.tmp.WERInternalMetadata.xml
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER.979928cc-515a-4a48-b19e-1fb6c14e935e.tmp.csv
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER.93971f8e-5076-4e67-b37f-41e9c7d8d3f9.tmp.txt
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER.38eff5e8-1695-4858-a7c8-d7e56203661f.tmp.xml

These files may be available here:
\\?\C:\ProgramData\Microsoft\Windows\WER\ReportArchive\AppCrash_96ba052f-0948-44_4c4e51f59fee85acf7cfcf241164bed5b2286_db8b36c1_a7cb4ac1-e806-44a2-bf3a-53ef9a6436c3

Analysis symbol: 
Rechecking for solution: 0

Report Id: 2dab20ca-62b8-459e-a36d-9ccd3d6a2236
Report Status: 268435456
Hashed bucket: b264bd9a4f5548fc681b07c75685caeb
Cab Guid: 0
```
[Report.wer.txt](https://github.com/benaclejames/VRCFaceTracking/files/11782326/Report.wer.txt)

benaclejames, 2023-06-18:
These are not `latest.log`s. You can find your latest logs in `%appdata%/VRCFaceTracking/latest.log`

AmaliaKalio, 2023-06-18:
> These are not `latest.log`s. You can find your latest logs in `%appdata%/VRCFaceTracking/latest.log`

This file appears to be empty.
[latest.log](https://github.com/benaclejames/VRCFaceTracking/files/11782388/latest.log)

benaclejames, 2023-06-18:
Have you tried wiping that whole VRCFaceTracking directory within appdata and then re-running?

AmaliaKalio, 2023-06-18:
> Have you tried wiping that whole VRCFaceTracking directory within appdata and then re-running?

That fixed it!

benaclejames, 2023-06-18:
*huh*, don't suppose you have a copy of what was there so I can diagnose for the future?

AmaliaKalio, 2023-06-18:
Wish I had backed it up. **However.** I tried a clean install of 5.0.2.1 and stepped through the reset buttons to try and replicate where I was at the start of this thread. The program would no longer launch, but the directory in %appdata% was also missing, strangely enough.

benaclejames, 2023-06-21:
> I had the same problem and I also solved it only after deleting the %appdata%/VRCFaceTracking folder. @benaclejames If you want a zip file of the contents please let me know and I will send it to you via Discord.

Yes please! This would really help in figuring out why the crash was happening in the first place. I can be reached on discord with the username `benaclejames`

Codel1417, 2023-07-01:
experiencing this now, No folder in %appdata% for VRCFaceTracking

`Faulting application name: VRCFaceTracking.exe, version: 5.0.2.3, time stamp: 0x641de100
Faulting module name: Microsoft.ui.xaml.dll, version: 3.0.0.2301, time stamp: 0x655f8cbf
Exception code: 0xc000027b
Fault offset: 0x0000000000341e5a
Faulting process id: 0x0x7D2C
Faulting application start time: 0x0x1D9ABB303F906CB
Faulting application path: C:\Program Files\WindowsApps\96ba052f-0948-44d8-86c4-a0212e4ae047_5.0.2.3_x64__d7rcq4vxghz0r\VRCFaceTracking.exe
Faulting module path: C:\Program Files\WindowsApps\96ba052f-0948-44d8-86c4-a0212e4ae047_5.0.2.3_x64__d7rcq4vxghz0r\Microsoft.ui.xaml.dll
Report Id: d9dee411-457f-4dcf-9afc-a70966a25bff
Faulting package full name: 96ba052f-0948-44d8-86c4-a0212e4ae047_5.0.2.3_x64__d7rcq4vxghz0r
Faulting package-relative application ID: App`

Codel1417, 2023-07-02:
It may be a permission issue as giving the Users group access to the appdata/VRCFaceTracking folder caused the log file to work

dfgHiatus, 2025-07-24:
Hello, please download the latest version of VRCFaceTracking from [Steam](https://store.steampowered.com/app/3329480/VRCFaceTracking/) to resolve this.

closed 2025-07-24 by dfgHiatus