benaclejames/VRCFaceTracking #153: 5.0.2.1 crashes immediately upon loading the quest pro module

sgao2022, 2023-06-13:


Gameruler02, 2023-06-13:
same with the sranipal tracking module

deadcxap, 2023-06-14:
Likewise, even after a clean install, the application terminates instantly. The window doesn't even appear.
Checked versions 5.0.2.2, 5.0.2.1, 5.0.2.0. The last one running is 5.0.1.1.
Tried deleting all modules, clearing application data.

benaclejames, 2023-06-14:
Have you tried deleting the current install, then downloading 5.0.2.2 using the .msix in the latest release?

AmaliaKalio, 2023-06-14:
> Likewise, even after a clean install, the application terminates instantly. The window doesn't even appear. Checked versions 5.0.2.2, 5.0.2.1, 5.0.2.0. The last one running is 5.0.1.1. Tried deleting all modules, clearing application data.

I believe this is the same as [Issue 152](https://github.com/benaclejames/VRCFaceTracking/issues/152)

benaclejames, 2023-06-17:
Update 5.0.2.3 is now out. Same issue?

sgao2022, 2023-06-18:
> 

It is resolved. Thank you and other contributors for the fix!

closed 2023-06-18 by sgao2022