benaclejames/VRCFaceTracking #154: Socket issue. "A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond."

UwUo-oUwU, 2023-06-20:
Log:
[VRCFaceTracking.Services.ActivationService] Information: VRCFT Version 5.0.2.3 initializing...
[VRCFaceTracking.Services.ActivationService] Information: Initializing OSC...
[VRCFaceTracking.Services.ActivationService] Information: Initializing main service...
[VRCFaceTracking.Core.Services.ModuleInstaller] Information: Uninstalled module 74f18332-1bcf-4b55-832d-9c91069afa31 from C:\Users\corey\AppData\Roaming\VRCFaceTracking\CustomLibs\74f18332-1bcf-4b55-832d-9c91069afa31
[VRCFaceTracking.Services.ActivationService] Information: Checking for updates for installed modules...
[VRCFaceTracking.Services.ActivationService] Information: Initializing modules...
[UnifiedLibManager] Information: Starting initialization tracking
[UnifiedLibManager] Information: Tearing down all modules...
[UnifiedLibManager] Information: Initializing runtimes...
[UnifiedLibManager] Information: Initializing Pico4SAFTExtTrackingModule, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null
[UnifiedLibManager] Information: Loading External Module Pico4SAFTExtTrackingModule, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null
[Pico4SAFTExtTrackingModule] Information: Initializing Pico Streaming Assistant data stream.
[Pico4SAFTExtTrackingModule] Information: System.Net.Sockets.SocketException (10060): A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond.
   at System.Net.Sockets.Socket.ReceiveFrom(Byte[] buffer, Int32 offset, Int32 size, SocketFlags socketFlags, EndPoint& remoteEP)
   at System.Net.Sockets.UdpClient.Receive(IPEndPoint& remoteEP)
   at Pico4SAFTExtTrackingModule.Pico4SAFTExtTrackingModule.ReceivePxrData()
[UnifiedLibManager] Warning: No modules loaded.
[VRCFaceTracking.ViewModels.RiskySettingsViewModel] Information: Reinitializing VRCFT...
[MainStandalone] Information: VRCFT Standalone Exiting!
[UnifiedLibManager] Information: Tearing down all modules...
[VRCFaceTracking.ViewModels.RiskySettingsViewModel] Information: Resetting VRCFT...
[VRCFaceTracking.ViewModels.RiskySettingsViewModel] Information: Resetting VRChat avatar configuration...
[MainStandalone] Warning: Native tracking parameters detected.
[MainStandalone] Information: 2 parameters loaded.
[MainStandalone] Warning: Native tracking parameters detected.
[MainStandalone] Information: 2 parameters loaded.
[MainStandalone] Information: 18 parameters loaded.
[MainStandalone] Warning: 18 Legacy parameters detected. Please consider updating the avatar to use the latest documented parameters.
[MainStandalone] Information: 18 parameters loaded.
[MainStandalone] Warning: 18 Legacy parameters detected. Please consider updating the avatar to use the latest documented parameters.
[VRCFaceTracking.ViewModels.RiskySettingsViewModel] Information: Reinitializing VRCFT...
[MainStandalone] Information: VRCFT Standalone Exiting!
[UnifiedLibManager] Information: Tearing down all modules...
--------------------------------------------------------------------------
Any advice to remedy this issue? Pico 4 Pro, latest software & Streaming Assistant which is OSC tracking capable. I reset all as well. Thanks ahead of time!

UwUo-oUwU, 2023-06-24:
Solved: Functioning fork for "PicoStreamingAssistantFTUDP" that solves this issue is "https://github.com/rogermiranda1000/PicoStreamingAssistantFTUDP/releases" by rogermiranda1000. The listed one is out of date in VRCFT.

UwUo-oUwU, 2023-07-13:
Please consider this updated module. I am re-opening due inoperable one still being used. Change log here: https://github.com/rogermiranda1000/PicoStreamingAssistantFTUDP/compare/1.2.0...1.3.0 
Previous (current on VRFT) is outdated & ports are closed, application unable to read face/eye data. Face tracking will not work without this update for the Pico 4 Pro/Enterprise. https://github.com/rogermiranda1000/PicoStreamingAssistantFTUDP/releases

rogermiranda1000, 2024-03-17:
@UwUo-oUwU the module was updated on the official repo

closed 2024-04-16 by benaclejames

Hatsunefuture, 2025-02-09:
[VRCFaceTracking.Services.ActivationService] Information: VRCFT Version 5.2.3.0 initializing...
[VRCFaceTracking.Services.ActivationService] Information: Initializing OSC...
[VRCFaceTracking.Core.OSC.OscQueryService] Debug: OSC Service Initializing
[VRCFaceTracking.Core.OSC.OscQueryService] Debug: OSC Service Initialized with result (False, False)
[VRCFaceTracking.Services.ActivationService] Information: Initializing main service...
[VRCFaceTracking.Core.OSC.OscQueryService] Information: OSCQuery detected. Setting port negotiation to autopilot.
[VRCFaceTracking.Core.Params.Data.UnifiedTrackingMutator] Debug: Reading configuration...
[VRCFaceTracking.Core.MainStandalone] Debug: Starting OSC update loop...
[VRCFaceTracking.Services.ActivationService] Debug: Checking for deletion requests for installed modules...
[VRCFaceTracking.Services.ActivationService] Information: Checking for updates for installed modules...
[VRCFaceTracking.Core.mDNS.MulticastDnsService] Debug: Advertising service: _oscjson._tcp, instance: VRCFT-74ZQ5X, port: 58063, address: 127.0.0.1
[VRCFaceTracking.Core.mDNS.MulticastDnsService] Debug: Advertising service: _osc._udp, instance: VRCFT-74ZQ5X, port: 53019, address: 127.0.0.1
[VRCFaceTracking.Core.Params.Data.UnifiedTrackingMutator] Debug: Configuration loaded.
[VRCFaceTracking.Core.OscQueryConfigParser] Debug: {"FULL_PATH":"/avatar","ACCESS":0,"CONTENTS":{"change":{"DESCRIPTION":"Avatar ID, updated whenever the user switches into a valid avatar.","FULL_PATH":"/avatar/change","ACCESS":1,"TYPE":"s","VALUE":["avtr_dba80bbb-a318-4d67-a608-a1f974ab24b7"]},"parameters":{"FULL_PATH":"/avatar/parameters","ACCESS":0,"CONTENTS":{"VelocityX":{"FULL_PATH":"/avatar/parameters/VelocityX","ACCESS":1,"TYPE":"f","VALUE":[0.0]},"VelocityY":{"FULL_PATH":"/avatar/parameters/VelocityY","ACCESS":1,"TYPE":"f","VALUE":[0.0]},"VelocityZ":{"FULL_PATH":"/avatar/parameters/VelocityZ","ACCESS":1,"TYPE":"f","VALUE":[0.0]},"AngularY":{"FULL_PATH":"/avatar/parameters/AngularY","ACCESS":1,"TYPE":"f","VALUE":[0.0]},"Grounded":{"FULL_PATH":"/avatar/parameters/Grounded","ACCESS":1,"TYPE":"T","VALUE":[true]},"AFK":{"FULL_PATH":"/avatar/parameters/AFK","ACCESS":1,"TYPE":"T","VALUE":[false]},"Upright":{"FULL_PATH":"/avatar/parameters/Upright","ACCESS":1,"TYPE":"f","VALUE":[0.4801072]},"TrackingType":{"FULL_PATH":"/avatar/parameters/TrackingType","ACCESS":1,"TYPE":"i","VALUE":[3]},"VRMode":{"FULL_PATH":"/avatar/parameters/VRMode","ACCESS":1,"TYPE":"i","VALUE":[1]},"MuteSelf":{"FULL_PATH":"/avatar/parameters/MuteSelf","ACCESS":1,"TYPE":"T","VALUE":[false]},"Voice":{"FULL_PATH":"/avatar/parameters/Voice","ACCESS":1,"TYPE":"f","VALUE":[0.0]},"Earmuffs":{"FULL_PATH":"/avatar/parameters/Earmuffs","ACCESS":1,"TYPE":"T","VALUE":[false]},"VelocityMagnitude":{"FULL_PATH":"/avatar/parameters/VelocityMagnitude","ACCESS":1,"TYPE":"f","VALUE":[0.0]},"ScaleFactor":{"FULL_PATH":"/avatar/parameters/ScaleFactor","ACCESS":1,"TYPE":"f","VALUE":[1.0]},"ScaleFactorInverse":{"FULL_PATH":"/avatar/parameters/ScaleFactorInverse","ACCESS":1,"TYPE":"f","VALUE":[1.0]},"ScaleModified":{"FULL_PATH":"/avatar/parameters/ScaleModified","ACCESS":1,"TYPE":"T","VALUE":[false]},"Bag":{"FULL_PATH":"/avatar/parameters/Bag","ACCESS":0,"CONTENTS":{"Lid_Check_IsGrabbed":{"FULL_PATH":"/avatar/parameters/Bag/Lid_Check_IsGrabbed","ACCESS":3,"TYPE":"T","VALUE":[false]},"Lid_Check_IsPosed":{"FULL_PATH":"/avatar/parameters/Bag/Lid_Check_IsPosed","ACCESS":3,"TYPE":"T","VALUE":[false]},"Lid_Check_Angle":{"FULL_PATH":"/avatar/parameters/Bag/Lid_Check_Angle","ACCESS":3,"TYPE":"f","VALUE":[0.0]},"Lid_Check_Stretch":{"FULL_PATH":"/avatar/parameters/Bag/Lid_Check_Stretch","ACCESS":3,"TYPE":"f","VALUE":[0.0]},"Lid_Check_Squish":{"FULL_PATH":"/avatar/parameters/Bag/Lid_Check_Squish","ACCESS":3,"TYPE":"f","VALUE":[0.0]}}},"pcs":{"FULL_PATH":"/avatar/parameters/pcs","ACCESS":0,"CONTENTS":{"contact":{"FULL_PATH":"/avatar/parameters/pcs/contact","ACCESS":0,"CONTENTS":{"intensity":{"FULL_PATH":"/avatar/parameters/pcs/contact/intensity","ACCESS":3,"TYPE":"i","VALUE":[0]},"in":{"FULL_PATH":"/avatar/parameters/pcs/contact/in","ACCESS":3,"TYPE":"T","VALUE":[false]},"out":{"FULL_PATH":"/avatar/parameters/pcs/contact/out","ACCESS":3,"TYPE":"T","VALUE":[false]},"hit":{"FULL_PATH":"/avatar/parameters/pcs/contact/hit","ACCESS":3,"TYPE":"T","VALUE":[false]},"proximity":{"FULL_PATH":"/avatar/parameters/pcs/contact/proximity","ACCESS":3,"TYPE":"f","VALUE":[0.0]},"enterPass":{"FULL_PATH":"/avatar/parameters/pcs/contact/enterPass","ACCESS":3,"TYPE":"f","VALUE":[0.0]},"handTouch":{"FULL_PATH":"/avatar/parameters/pcs/contact/handTouch","ACCESS":3,"TYPE":"f","VALUE":[0.0]},"velocity-in":{"FULL_PATH":"/avatar/parameters/pcs/contact/velocity-in","ACCESS":3,"TYPE":"f","VALUE":[0.0]},"velocity-out":{"FULL_PATH":"/avatar/parameters/pcs/contact/velocity-out","ACCESS":3,"TYPE":"f","VALUE":[0.0]},"detect-in":{"FULL_PATH":"/avatar/parameters/pcs/contact/detect-in","ACCESS":3,"TYPE":"T","VALUE":[false]},"detect-out":{"FULL_PATH":"/avatar/parameters/pcs/contact/detect-out","ACCESS":3,"TYPE":"T","VALUE":[false]},"detect-hit":{"FULL_PATH":"/avatar/parameters/pcs/contact/detect-hit","ACCESS":3,"TYPE":"T","VALUE":[false]},"velocity-smash":{"FULL_PATH":"/avatar/parameters/pcs/contact/velocity-smash","ACCESS":3,"TYPE":"f","VALUE":[0.0]}}},"isEnable":{"FULL_PATH":"/avatar/parameters/pcs/isEnable","ACCESS":3,"TYPE":"T","VALUE":[false]},"reset":{"FULL_PATH":"/avatar/parameters/pcs/reset","ACCESS":3,"TYPE":"T","V [... cut ...]
   at System.Net.Http.ConnectHelper.EstablishSslConnectionAsync(SslClientAuthenticationOptions sslOptions, HttpRequestMessage request, Boolean async, Stream stream, CancellationToken cancellationToken)
[App] Critical: Message: The SSL connection could not be established, see inner exception.
[VRCFaceTracking.Core.OSC.Query.mDNS.HttpHandler] Debug: Responding to oscquery host info request with {"NAME":"VRCFT-74ZQ5X","EXTENSIONS":{"ACCESS":true,"CLIPMODE":false,"RANGE":true,"TYPE":true,"VALUE":true},"OSC_IP":"127.0.0.1","OSC_PORT":53019,"OSC_TRANSPORT":"UDP"}
[VRCFaceTracking.Services.OpenVRService] Information: Successfully initialized OpenVR