benaclejames/VRCFaceTracking #161: Error opening Quest OpenXR

keithdefox, 2023-08-04:
So I have recently got a meta quest pro and tried to follow along tutorials for activating face and eye tracking. I followed the steps properly, everything worked, until I tried to download the program. I installed the module but found my headset never popped up, and the app didn't register on steamvr or quest link, despite it being set up and active at the time. When looking at the console, I saw this: 

[UnifiedLibManager] Debug: QuestProTrackingModule properly implements ExtTrackingModule.
[UnifiedLibManager] Debug: Initializing requested runtimes...
[UnifiedLibManager] Information: Initializing runtimes...
[UnifiedLibManager] Information: Initializing VRCFT - Quest OpenXR, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null
[UnifiedLibManager] Information: Loading External Module VRCFT - Quest OpenXR, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null
[QuestProTrackingModule] Error: [QuestOpenXR] Failed to create XrInstance.
[QuestProTrackingModule] Error: [QuestOpenXR] Please make sure that the Oculus application is running, or this module did not properly uninitialize.
[UnifiedLibManager] Warning: No modules loaded.

I couldn't find a single documented case of this error. I have enabled "Oculus is set as the active OpenXR runtime" what might be happening? And how the hell do I fix it? Right now my quest is a brick and I'm freaking out. 

Thanks for the help, if you can.

keithdefox, 2023-08-04:
I think Steamvr wasn't running and that was the issue. I might revisit this if I find the problem continues. For some reason the OpenXR thingy was buggy earlier.

closed 2023-08-04 by keithdefox

benaclejames, 2023-08-04:
> I couldn't find a single documented case of this error
Then you didn't take more than 10 minutes searching the discord or the docs. Both have a search box, both would take you to the solution had you typed in "xrinstance".

keithdefox, 2023-08-04:
Sorry about that. I'd been working all afternoon on setting everything up and did a general search on Google and couldn't find anything popping up for the code. I had a panic attack and was freaking out after bashing my head against the wall for an hour. Thank you for the help, I really appreciate it. You have a cool app, can't wait to try it out