benaclejames/VRCFaceTracking #162: The Install module button does nothing

orcinus, 2023-08-04:
Reproduction steps:
1) click "Install some from here" from the "No Modules Installed" notification
2) click "Install" button for SRanipalTrackingModule
3) nothing happens

Expected result:
Module gets downloaded and installed.

Version: 5.1.1.0

Platform: Windows 10 22H2

orcinus, 2023-08-04:
Log output:

```
[UnifiedLibManager]` Information: Starting initialization tracking
[UnifiedLibManager] Information: Tearing down all modules...
[UnifiedLibManager] Debug: Initializing requested runtimes...
[UnifiedLibManager] Information: Initializing runtimes...
[UnifiedLibManager] Warning: No modules loaded.
[VRCFaceTracking.Core.Services.ModuleDataService] Debug: Rating for 74f18332-1bcf-4b55-832d-9c91069afa31 was not found
[UnifiedLibManager] Information: Tearing down all modules...
[VRCFaceTracking.Core.Services.ModuleInstaller] Debug: Uninstalling module 74f18332-1bcf-4b55-832d-9c91069afa31
[VRCFaceTracking.Core.Services.ModuleInstaller] Warning: Module 74f18332-1bcf-4b55-832d-9c91069afa31 could not be found where it was expected in C:\Users\Ante Vukorepa\AppData\Roaming\VRCFaceTracking\CustomLibs\74f18332-1bcf-4b55-832d-9c91069afa31
[App] Error: Unhandled exception
[App] Critical: Stacktrace:    at System.IO.FileSystem.MoveDirectory(String sourceFullPath, String destFullPath, Boolean isCaseSensitiveRename)
   at System.IO.FileSystem.MoveDirectory(String sourceFullPath, String destFullPath)
   at VRCFaceTracking.Core.Services.ModuleInstaller.InstallRemoteModule(TrackingModuleMetadata moduleMetadata)
   at VRCFaceTracking.Views.ModuleRegistryDetailControl.Install_Click(Object sender, RoutedEventArgs e)
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__128_0(Object state)
   at Microsoft.UI.Dispatching.DispatcherQueueSynchronizationContext.<>c__DisplayClass2_0.<Post>b__0()
[App] Critical: Inner exception: (null)
[App] Critical: Message: Source and destination path must have identical roots. Move will not work across volumes.
[UnifiedLibManager] Information: Tearing down all modules...
```

orcinus, 2023-08-04:
This appears to be because the app assumes it's on C:\
If it's not (and it can not-be, because Microsoft allows you to change the default App drive via Settings), a lot of things will break.

orcinus, 2023-08-04:
Nope. Uninstalled, reconfigured app storage back to C:, reinstalled, same issue persists.
Also, no VRCFaceTracking folder in AppData.
No idea what's going on.

orcinus, 2023-08-04:
Okay, figured it out.
App expects the Windows Temp folder to be on the same driver as itself.

Since it uses move instead of copy (and delete), which only works within the same volume, copying the module from temp where it was downloaded, to where it needs to be, fails if Temp is elsewhere.

dd1231512dd, 2023-10-03:
How can I fix it?

Charmandrigo, 2024-01-06:
> Nope. Uninstalled, reconfigured app storage back to C:, reinstalled, same issue persists.
> 
> Also, no VRCFaceTracking folder in AppData.
> 
> No idea what's going on.

This is the issue I been getting except I can wipe the folders within app data and the built-in modules load.

A second issue is that no matter what I do, the 3rd party modules (like eyetrackvr or pimax) will never load, not even if I install them manually as a zip, modules are added to the app data folder, but these fail to load completely.

To the devs, I did a more specific rundown on my thread on discord: https://discord.com/channels/849300336128032789/1189778058610233404

I should mention I am on windows 11 21h3, so please check if you are able to recreate this issue on the most recent w11 version.
So far, un installing, reinstalling, wiping data etc has not worked. Not even adding the modules directly on CustomLibs, it's become clear that only the modules located on StockLibs are the only ones that load correctly on the app

dfgHiatus, 2025-07-24:
Closing this issue due to inactivity.

closed 2025-07-24 by dfgHiatus