benaclejames/VRCFaceTracking #163: Installing SRanipalTrackingModule without SRanipal Runtime causes a crash

orcinus, 2023-08-04:
Reproduction steps:

1) uninstall SRanipal Runtime
2) click "Install some from here" from the "No Modules Installed" notification
3) click "Install" button for SRanipalTrackingModule
4) application crashes

Expected result:
Display an error prompting to install SRanipal runtime, if that's the cause of the crash.
Otherwise, no crash and no error.

Version: 5.1.1.0

Platform: Windows 10 22H2

orcinus, 2023-08-04:
Additionally:

Yes, i am aware the instructions state "Don't forget to install SRanipal Runtime before installing this module".
But that should be displayed as an error / prompt, not buried inside the module metadata. That is just silly from a UX POV.

benaclejames, 2023-08-04:
Hmm, not sure how to implement this from a UI design perspective. Please feel free to PR whatever you think necessary :)

orcinus, 2023-08-04:
Handling an unhandled exception and popping up an error would probably be enough.

orcinus, 2023-08-04:
Hmmm. Actually, nevermind, this seems to be unrelated to SRanipal missing.
It just happened with it installed and running.