benaclejames/VRCFaceTracking #166: VRCFaceTracking stuck on loading with Quest Pro

Chekote, 2023-08-10:
Howdy!

First, thanks so much for making this software. You're awesome!

I'm not sure if this is the correct place to post this, so please let me know if I should post it elsewhere. Thanks :)

## The Problem

Unfortunately, I am not having much luck getting it to work. I've followed the documentation [here](https://docs.vrcft.io/docs/hardware/quest-pro), and the video [here](https://www.youtube.com/watch?v=Keh__Lw7dEk&t=3s), but I am having a problem where when I launch the VRCFaceTracking app from my Quest Pro, it gets stuck on this screen:

[image]

According to [the video](https://www.youtube.com/watch?v=Keh__Lw7dEk&t=3s), I should see a splash screen of the VRCFaceTracking app, not this weird looking loading screen.

Additionally, and probably caused by the same root issue, is that the Avatar Status section of VRCFaceTracking always shows:

```
Loading...
Loading...
```

Does anyone know why this is happened, and what I need to do to fix it?

## Troubleshooting so far:
* I have re-read the docs and re-watched the video multiple times, double checking that I did everything as explained.
* I have restarted both my PC and the Quest Pro
* I have tried both with cabled Oculus Link and Air Link
* I have re-installed VRCFaceTracking
* I have disabled and re-enabled developer mode on the Quest Pro

## My Setup

### VRC Face Tracking
I am using [VRCFaceTracking 5.1.1.0](https://github.com/benaclejames/VRCFaceTracking/tree/5.1.1.0).

I have the Quest Pro OpenXR v1.1 module installed:
[image]

The VRCFaceTracking app is showing that it sees the Quest Pro and that the Eye and Face tracking are both active:
[image]

## Oculus Desktop App
I have set Oculus as the default OpenXR Runtime as per the video:
[image]

The Oculus Desktop App Beta settings are as follows:
[image]

## Quest Pro
I have enabled both the Natural Facial Expressions and the Eye Tracking features in the Quest Pro itself and confirmed that they are working via the homeworld mirror.

KazyEXE, 2023-08-10:
Press the oculus button, switch back over to SteamVR.

Chekote, 2023-08-10:
@KazyEXE thanks for the reply. I guess I wasn't clear. I have added additional info to the original post above to make it clear that I know how to get into VRC:\

> I then switch over to SteamVR, launch VRChat and use any avatar with face tracking from Jerry's Mod world, it does not work. No eye tracking, no blinking, no face tracking, nothing.

InconsolableCellist, 2023-08-10:
Did you enable eye and lip tracking in the radial menu while in the avatar? If you open OSC Debug in the radial menu do you see parameters being updated? And in VRCFT at that point, does it ever detect the avatar being loaded? Do you use any other OSC applications or have any command line args to VRC?

kusomaigo, 2023-08-10:
This is such a beautifully detailed issue post that just, was posted in the wrong place..
My guess is OSC is probably disabled in VRChat or you're having other OSC related issues.
Next time use the Discord for problems like this, and make an issue here when it is a code issue with the program, not a module or setup problem!

Chekote, 2023-08-10:
@InconsolableCellist thanks for the reply 🙇‍♂️ 

> If you open OSC Debug in the radial menu do you see parameters being updated?
I was not aware that OSC menu existed! I checked it out and yes, the parameters are being displayed. Since you clued me in to the existence of this I took a look at the other options. Disabling and re-enabling OSC here fixed the issue! Thank you so much!!! ❤️

Chekote, 2023-08-10:
@kusomaigo thanks for the kind words. I will hit up Discord with any further issues. :)

closed 2023-08-10 by Chekote