benaclejames/VRCFaceTracking #168: very bad bug

ghostie5, 2023-08-11:
I have been using this software a lot, and i have been having a really bad bug as of today that isnt being fixed for me at all. my face tracker will not pick up anything but 0m/s the entire time im using it. but the catch is it is only on vrcft. it works on neos, and works with motion capture. but vrcft randomly wont pick it up. please fix this

ghostie5, 2023-08-12:
> Does your issue also include your modules not loading at ALL, because I've been having the same issue on my end and I JUST got my face 

it loads, that would be a firmware issue I belive with srinepal. but mine is just the software glitching

benaclejames, 2023-12-24:
Is this resolved?

closed 2024-02-12 by benaclejames