benaclejames/VRCFaceTracking #174: Freezes (white screen) on launch

i0ntempest, 2023-09-28:
The app will freeze on launch, and after a while it will open with the top (loaded module) part missing. This happens almost every time when I want to use VRCFT. I have to try multiple times launching the app to get it to load a module. Apparently it's failing to check module updates, but that shouldn't freeze the whole app?
Output log:
```
[MainStandalone] Information: VRCFT Version 5.1.1.0 initializing...
[MainStandalone] Information: Initializing OSC...
[OSC] Debug: OSC Service Initializing
[OSC] Debug: Binding Sender Client
[OSC] Debug: OSC Service Initialized with result (True, True)
[MainStandalone] Information: Initializing main service...
[VRCFaceTracking.UnifiedTrackingMutator] Debug: Reading configuration...
[MainStandalone] Debug: Starting OSC update loop...
[MainStandalone] Debug: Checking for deletion requests for installed modules...
[MainStandalone] Information: Checking for updates for installed modules...
[App] Error: Unhandled exception
[App] Critical: Stacktrace:    at System.Threading.Tasks.Task.ThrowIfExceptional(Boolean includeTaskCanceledExceptions)
   at System.Threading.Tasks.Task`1.GetResultCore(Boolean waitCompletionNotification)
   at VRCFaceTracking.Core.Services.ModuleDataService.AllModules()
   at VRCFaceTracking.Core.Services.ModuleDataService.GetRemoteModules()
   at VRCFaceTracking.Services.ActivationService.StartupAsync()
   at VRCFaceTracking.Services.ActivationService.ActivateAsync(Object activationArgs)
   at VRCFaceTracking.App.OnLaunched(LaunchActivatedEventArgs args)
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__128_0(Object state)
   at Microsoft.UI.Dispatching.DispatcherQueueSynchronizationContext.<>c__DisplayClass2_0.<Post>b__0()
[App] Critical: Inner exception: System.Threading.Tasks.TaskCanceledException: A task was canceled.
   at System.Threading.Tasks.Task.GetExceptions(Boolean includeTaskCanceledExceptions)
   at System.Threading.Tasks.Task.ThrowIfExceptional(Boolean includeTaskCanceledExceptions)
   at System.Threading.Tasks.Task`1.GetResultCore(Boolean waitCompletionNotification)
   at VRCFaceTracking.Core.Services.ModuleDataService.AllModules()
   at VRCFaceTracking.Core.Services.ModuleDataService.GetRemoteModules()
   at System.Runtime.CompilerServices.AsyncMethodBuilderCore.Start[TStateMachine](TStateMachine& stateMachine)
   at VRCFaceTracking.Core.Services.ModuleDataService.GetRemoteModules()
   at VRCFaceTracking.Services.ActivationService.StartupAsync()
   at System.Runtime.CompilerServices.AsyncMethodBuilderCore.Start[TStateMachine](TStateMachine& stateMachine)
   at VRCFaceTracking.Services.ActivationService.StartupAsync()
   at VRCFaceTracking.Services.ActivationService.ActivateAsync(Object activationArgs)
   at System.Runtime.CompilerServices.AsyncTaskMethodBuilder`1.AsyncStateMachineBox`1.ExecutionContextCallback(Object s)
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state)
   at System.Runtime.CompilerServices.AsyncTaskMethodBuilder`1.AsyncStateMachineBox`1.MoveNext(Thread threadPoolThread)
   at System.Runtime.CompilerServices.AsyncTaskMethodBuilder`1.AsyncStateMachineBox`1.MoveNext()
   at Microsoft.UI.Dispatching.DispatcherQueueSynchronizationContext.<>c__DisplayClass2_0.<Post>b__0()
   at ABI.Microsoft.UI.Dispatching.DispatcherQueueHandler.Do_Abi_Invoke(IntPtr thisPtr)
   at ABI.Microsoft.UI.Xaml.IApplicationStaticsMethods.Start(IObjectReference _obj, ApplicationInitializationCallback callback)
   at Microsoft.UI.Xaml.Application.Start(ApplicationInitializationCallback callback)
   at VRCFaceTracking.Program.Main(String[] args)
--- End of stack trace from previous location ---

[App] Critical: Message: One or more errors occurred. (A task was canceled.)
[VRCFaceTracking.UnifiedTrackingMutator] Debug: Configuration loaded.
```

InconsolableCellist, 2023-09-29:
This sounds like a network-level issue with fetching updates. Can you try on a VPN?

i0ntempest, 2023-09-29:
> This sounds like a network-level issue with fetching updates. Can you try on a VPN?

I'm thinking that as well, is I'm using a 6in4 tunnel and I bet that's causing problems. Can I have the domains it will connect to for updates so I can set them to not resolve to IPv6 in my DNS server?

InconsolableCellist, 2023-09-29:
Try checking: https://rjlk4u22t36tvqz3bvbkwv675a0wbous.lambda-url.us-east-1.on.aws/modules

i0ntempest, 2023-09-30:
Thanks, apparently AWS hates my tunnel. Connecting through IPv4 works fine.
You may close this now, but as I said a failed update check really shouldn't affect loading installed modules.

closed 2023-09-30 by InconsolableCellist