benaclejames/VRCFaceTracking #175: Module Install button not working

dd1231512dd, 2023-10-04:
In Module Registry, Install button not working.
here is error log

[MainStandalone] Information: VRCFT Version 5.1.1.0 initializing...
[MainStandalone] Information: Initializing OSC...
[OSC] Debug: OSC Service Initializing
[OSC] Debug: Binding Sender Client
[OSC] Debug: OSC Service Initialized with result (True, True)
[MainStandalone] Information: Initializing main service...
[VRCFaceTracking.UnifiedTrackingMutator] Debug: Reading configuration...
[MainStandalone] Debug: Starting OSC update loop...
[MainStandalone] Debug: Checking for deletion requests for installed modules...
[MainStandalone] Information: Checking for updates for installed modules...
[MainStandalone] Information: Initializing modules...
[UnifiedLibManager] Information: Starting initialization tracking
[UnifiedLibManager] Information: Tearing down all modules...
[UnifiedLibManager] Debug: Initializing requested runtimes...
[UnifiedLibManager] Information: Initializing runtimes...
[UnifiedLibManager] Warning: No modules loaded.
[VRCFaceTracking.UnifiedTrackingMutator] Debug: Configuration loaded.
[VRCFaceTracking.Core.Services.ModuleDataService] Debug: Rating for 74f18332-1bcf-4b55-832d-9c91069afa31 was not found
[UnifiedLibManager] Information: Tearing down all modules...
[VRCFaceTracking.Core.Services.ModuleInstaller] Debug: Uninstalling module e90de1b0-1d67-4d5b-b517-597043fa3852
[VRCFaceTracking.Core.Services.ModuleInstaller] Warning: Module e90de1b0-1d67-4d5b-b517-597043fa3852 could not be found where it was expected in C:\Users\user\AppData\Roaming\VRCFaceTracking\CustomLibs\e90de1b0-1d67-4d5b-b517-597043fa3852
[VRCFaceTracking.Core.Services.ModuleDataService] Debug: Rating for e90de1b0-1d67-4d5b-b517-597043fa3852 was not found
[App] Error: Unhandled exception
[App] Critical: Stacktrace:    at System.IO.FileSystem.MoveDirectory(String sourceFullPath, String destFullPath, Boolean isCaseSensitiveRename)
   at System.IO.FileSystem.MoveDirectory(String sourceFullPath, String destFullPath)
   at VRCFaceTracking.Core.Services.ModuleInstaller.InstallRemoteModule(TrackingModuleMetadata moduleMetadata)
   at VRCFaceTracking.Views.ModuleRegistryDetailControl.Install_Click(Object sender, RoutedEventArgs e)
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__128_0(Object state)
   at Microsoft.UI.Dispatching.DispatcherQueueSynchronizationContext.<>c__DisplayClass2_0.<Post>b__0()
[App] Critical: Inner exception: (null)
**[App] Critical: Message: Source and destination path must have identical roots. Move will not work across volumes.**

seascape, 2023-10-05:
Same problem. I'm following the tutorial for Virtual Desktop but at the module-installing step clicking the "Install" button on the VD module's page does nothing. It's as if the button's not being clicked.

Windows 10 x64 with a Quest Pro

dfgHiatus, 2025-07-24:
Closing this issue due to inactivity.

closed 2025-07-24 by dfgHiatus