benaclejames/VRCFaceTracking #176: Vrchat crashes every time i open face tracking application.

GalaxyPlanetarium, 2023-10-14:
(Disclaimer: This could be something I'm doing that could be entirely wrong and just not realize it or it could be the fact I'm running everything on a laptop.) So, let me start off by explaining, I am using a quest 2, a vive facial tracker, and have virtual desktop. When I first installed the application, I also installed SRanipalTrackingModule and ALVR Module. I wasn't able to get anything work correctly. it would connect and receive incoming by never outgoing. that was until I installed ALXR Local Module and the module for virtual desktop. but here is the kicker, I would receive 1m/s outgoing and then Vrchat would crash as soon as you open VRCFaceTracking.

Laptop is a ASUS TUF GAMING F17. Don't know if that helps with anything.

kusomaigo, 2023-10-14:
If you're asking for troubleshooting help ask in the VRCFT discord.

ThatGuyThimo, 2024-10-06:
this might be related to an issue in your your settings file. go to ```AppData\Roaming\VRCFaceTracking\VRCFaceTracking\ApplicationData``` and delete the LocalSettings.json file

to get to appdata pres win + r and type %appdata%