benaclejames/VRCFaceTracking #177: No modules loaded

Insal1027, 2023-11-08:
Until yesterday, the VRCFT worked well.

However, modules do not load after Windows update(?)

I did the following tests

1.Quest pro reset
2.OpenXR Runtime check
2.VRCFT reset / reinstall
3.Developer mode check (mobile & pc)
4.Quest Pro Open XR module reinstall
5.Window reset

Facial works well on Quest Pro (before airlink)

[image]

Output log)
```
[MainStandalone] Information: VRCFT Version 5.1.1.0 initializing...
[MainStandalone] Information: Initializing OSC...
[OSC] Debug: OSC Service Initializing
[OSC] Debug: Binding Sender Client
[OSC] Debug: OSC Service Initialized with result (True, True)
[MainStandalone] Information: Initializing main service...
[MainStandalone] Warning: VRCFT detected OSC was disabled and automatically enabled it.
[VRCFaceTracking.UnifiedTrackingMutator] Debug: Reading configuration...
[MainStandalone] Debug: Starting OSC update loop...
[MainStandalone] Debug: Checking for deletion requests for installed modules...
[MainStandalone] Information: Checking for updates for installed modules...
[VRCFaceTracking.UnifiedTrackingMutator] Debug: Configuration loaded.
[MainStandalone] Information: Initializing modules...
[UnifiedLibManager] Information: Starting initialization tracking
[UnifiedLibManager] Information: Tearing down all modules...
[UnifiedLibManager] Debug: QuestProTrackingModule properly implements ExtTrackingModule.
[UnifiedLibManager] Debug: Initializing requested runtimes...
[UnifiedLibManager] Information: Initializing runtimes...
[UnifiedLibManager] Information: Initializing VRCFT - Quest OpenXR, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null
[UnifiedLibManager] Information: Loading External Module VRCFT - Quest OpenXR, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null
[QuestProTrackingModule] Information: [QuestOpenXR] Eye Tracking not supported.
[UnifiedLibManager] Warning: No modules loaded.
```

Banou26, 2024-02-07:
Issue just happened to me too. It worked completely fine yesterday.
```
[MainStandalone] Information: VRCFT Version 5.1.1.0 initializing...
[MainStandalone] Information: Initializing OSC...
[OSC] Debug: OSC Service Initializing
[OSC] Debug: Binding Sender Client
[OSC] Debug: OSC Service Initialized with result (True, True)
[MainStandalone] Information: Initializing main service...
[VRCFaceTracking.UnifiedTrackingMutator] Debug: Reading configuration...
[MainStandalone] Debug: Starting OSC update loop...
[MainStandalone] Debug: Checking for deletion requests for installed modules...
[MainStandalone] Information: Checking for updates for installed modules...
[MainStandalone] Information: Initializing modules...
[UnifiedLibManager] Information: Starting initialization tracking
[UnifiedLibManager] Information: Tearing down all modules...
[VRCFaceTracking.UnifiedTrackingMutator] Debug: Configuration loaded.
[UnifiedLibManager] Debug: QuestProTrackingModule properly implements ExtTrackingModule.
[UnifiedLibManager] Debug: Initializing requested runtimes...
[UnifiedLibManager] Information: Initializing runtimes...
[UnifiedLibManager] Information: Initializing VRCFT - Quest OpenXR, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null
[UnifiedLibManager] Information: Loading External Module VRCFT - Quest OpenXR, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null
[QuestProTrackingModule] Error: [QuestOpenXR] Failed to create Face Treacker.
[UnifiedLibManager] Warning: No modules loaded.
```

@Insal1027 Has it fixed itself with some time? Anything you tried that managed to fix the issue?

Banou26, 2024-02-07:
After noticing that my Meta Quest Link software updated to v62, I checked for my headset's firmware version which was 59. I've updated it to latest(v62) and it started working again.

closed 2024-02-12 by benaclejames